MrChromebox/scripts #333: remettre chromeOS sur chomebox sur lequel UIEFI 4.20.0 05/15/2023 #75791 a étè installé

WASH90, 2023-11-20:
bonjour
J'ai un CHROMEBOOK HP 14a que j'ai intaller une version UIEFI 4.20.0.05/15/2023
et j'aimerais bien remettre la version de base de chromeOS en effet j'ai installé windows sur le (chrombox) hp 14A
mais hélas !! je n'arrive pas a excuter flashroom a travers le script << cd; curl -LOk https://mrchromebox.tech/setup-kodi.sh && sudo bash setup-kodi.sh -k >> sur invitation de commade de UBUNTU il commence le téléchargement mais après j'ai un message d'erreur
ci-joint 
[image]

j'ai modifie les paramettre du noyau en ajoutant "iomem=relaxed " 
a travers     gedit /etc/default/grub j'ai enregistré 
aprés j'ai effectue la commande sudo update-grub 
ci joint
[image] 
j'ai redemarer et j'ai relace la commade  << cd; curl -LOk https://mrchromebox.tech/setup-kodi.sh && sudo bash setup-kodi.sh -k >>
j'ai toujours le meme message d'erreur  
merçi d'avance ...

MrChromebox, 2023-11-20:
live boot Ubuntu 2304, install curl, run Firmware Utility script nothing else is needed.

Not sure why you're trying to run the deprecated Kodi setup script.

closed 2023-11-20 by MrChromebox

WASH90, 2023-11-21:
merci de la reponse....
j'ai fais une mise a jour a 23.04
j'ai intallé : curl et apres j'ai exercuté le script:" cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh"
helas j'ai le meme message d'erreur 

merçi d'avance 
[image]

MrChromebox, 2023-11-21:
not sure what the problem is and it's hard to work around using google translate

WASH90, 2023-11-23:
je ne comprend pas desolé.... comment si prendre avec **goole translate** soit plus explicite stp 
merçi d'avance

MrChromebox, 2023-11-23:
does `cat /proc/cmdline` show `iomem=relaxed` as one of the parameters?

WASH90, 2023-11-23:
oui il s'affiche comme l'un des parametre 
ci_joint la photo
[image]