MrChromebox/scripts #334: Low VRAM

Striker770, 2023-11-21:
Hello, 

Having Lenovo ThinkPad C13 Yoga Chromebook and installed the Mr.Chromebox firmware (Full UEFI). When i tried to install Windows, i saw that the VRAM is just 32MB. Verified it from AMD Control Panel. Is there any way to increase the VRAM?

Thanks

MrChromebox, 2023-11-21:
the VRAM is fixed at 32MB in the firmware and cannot be easily changed

Striker770, 2023-11-21:
> the VRAM is fixed at 32MB in the firmware and cannot be easily changed

Is it because of this Laptop or all the AMD (Ryzen) firmware from your side has 32MB limit? 

Any workaround for such case?

MrChromebox, 2023-11-21:
I would need to rebuild a custom AMD firmware component that is loaded by coreboot.

what is the impediment with only 32MB?

Striker770, 2023-11-21:
> I would need to rebuild a custom AMD firmware component that is loaded by coreboot.
> 
> what is the impediment with only 32MB?

I see. Got macOS installed on it. But it would need a minimum of 512MB VRAM. 2048MB would be great. I'm available on Discord or over email if you want me to do some tests and all.

MrChromebox, 2023-11-22:
reserving 2GB RAM for VRAM on an 8GB system is very hard to justify; 512MB isn't unreasonable.

> BTW, any specific reasons why you thought to use 32MB? If I'm not wrong your firmware is based on Coreboot right?

I didn't, Google did when they developed the firmware with AMD. My firmware is coreboot + edk2 payload.

Striker770, 2023-11-22:
> reserving 2GB RAM for VRAM on an 8GB system is very hard to justify; 512MB isn't unreasonable.

I thought the same initially but i have an HP Pavilion with everything stock (BIOS) and it has only 8GB RAM but still the VRAM is 2048MB which was way much surprising to me. 

> I didn't, Google did when they developed the firmware with AMD. My firmware is coreboot + edk2 payload.
Hmm. I see. Can we make some progress on this issue? 512MB VRAM would do the job! Let me know what do you need. Happy to contribute if you have any donation link.

Thanks again for this wonderful project!

MrChromebox, 2023-11-23:
> Hmm. I see. Can we make some progress on this issue? 512MB VRAM would do the job! Let me know what do you need. Happy to contribute if you have any donation link.

I'll try to do this over the weekend, a bit tied up with holiday stuff ATM. There's a paypal donate link on my site at the bottom of the nav menu ;-)

Striker770, 2023-11-23:
> > Hmm. I see. Can we make some progress on this issue? 512MB VRAM would do the job! Let me know what do you need. Happy to contribute if you have any donation link.
> 
> I'll try to do this over the weekend, a bit tied up with holiday stuff ATM. 

Sure. I'm excited. 

Happy Thanksgiving :)

> There's a paypal donate link on my site at the bottom of the nav menu ;-)

Thank you

MrChromebox, 2023-11-25:
@Striker770 please DM or ping me on discord

Striker770, 2023-11-27:
> @Striker770 please DM or ping me on discord

Done. Thank You

MrChromebox, 2023-11-27:
@Striker770 actually I just updated all of the Picasso images with a fix for the Athlon 3150C variants, and fixed 512MB VRAM as well. So just update from the script and see how that works for you

Striker770, 2023-11-27:
> @Striker770 actually I just updated all of the Picasso images with a fix for the Athlon 3150C variants, and fixed 512MB VRAM as well. So just update from the script and see how that works for you

OMG. Going to try it now and report. Meanwhile can you add me on the Discord please? Primeee#4056 is my username. 

Thanks again!

Striker770, 2023-11-27:
@MrChromebox So, i flashed using the new script using Ubuntu and i literally tried everything and there is no sign of display, nothing. The power LED flashes white and that's it. Seems like firmware got it broken. Any ways to recover?

Striker770, 2023-11-27:
@MrChromebox Okay, it's back to life. I can see the new version!

Reporting in a couple of minutes.

Striker770, 2023-11-27:
@MrChromebox And it's a SUCCESS!

Well done. Thank you so much. However, seems like it needs like 1GB VRAM for the proper operation as the system restarts abruptly.

MrChromebox, 2023-11-27:
@Striker770 then I'll need to do a one-off firmware for you, HMU in the AMD-picasso channel on the chrultrabook discord and I'll do that

Striker770, 2023-11-27:
> @Striker770 then I'll need to do a one-off firmware for you, HMU in the AMD-picasso channel on the chrultrabook discord and I'll do that

Can you please send me the invite? Not sure where i can find the link to join!

MrChromebox, 2023-11-27:
https://discord.gg/8ZwD4E3n

Striker770, 2023-11-27:
> https://discord.gg/8ZwD4E3n

Thank you!

closed 2023-11-28 by MrChromebox