MrChromebox/scripts #335: Lenovo Yoga C13 - Cannot update Firmware via firmware-util

al12gamer, 2023-12-05:
When running Fedora 39 via the UEFI Full ROM on the Lenovo Yoga C13 Chromebook, I seem to be unable to update the coreboot firmware on here.
Are there any suggestions for what to do when rebooting? It doesn't seem to take long to boot back up, but it apparently still has the same firmware.
[Here is an asciinema terminal recording](https://asciinema.org/a/SbvjflJqoQASlQIgYzDG8NKNT) of me updating the new firmware. I'll add another after I've rebooted.
Any ideas? Any log files I can send?

al12gamer, 2023-12-05:
Here's an [asciinema terminal recording ](https://asciinema.org/a/6C99GfCkcUDbnl8wdkj1iPsBt)after the reboot. The new firmware that was supposedly updated didn't take.

al12gamer, 2023-12-05:
...it also seems the trackpoint on here isn't working now

MrChromebox, 2023-12-05:
it's not the same firmware, and the trackpoint is disabled because it was breaking the keyboard on S3 resume under Windows

closed 2023-12-05 by MrChromebox

al12gamer, 2023-12-05:
Is there a way to keep it enabled on linux systems?

MrChromebox, 2023-12-05:
not currently, I'd need to build a separate image with it enabled. If you'd like that, ping me on Discord and I'll build it for you

al12gamer, 2023-12-05:
> not currently, I'd need to build a separate image with it enabled. If you'd like that, ping me on Discord and I'll build it for you

I'm unable to find a link to the Chrultrabook Discord at this time- is this because they were wanting people to move to the forum?

MrChromebox, 2023-12-05:
https://discord.gg/aKggq9Gt