MrChromebox/scripts #340: Morphius Lenovo ThinkPad C13 old firmware Trackpoint NoT working

bparihar, 2023-12-26:
i already flashed the old [morphius] firmware which a holding a bug for Trackpoint
now my Trackpoint ia not working can any one guide me with reflashing the new firmware on my C13 [morphius] I'm on Windows now 

Regards
Bhavesh

MrChromebox, 2023-12-26:
you cannot use the trackpoint-enabled firmware with Windows currently, due to incompatibilities with coolstar's Windows drivers

closed 2023-12-26 by MrChromebox

bparihar, 2023-12-26:
Can You guide me how can i reflash the right firmware again and from which Operating system i can do that .

Thankyou

bparihar, 2023-12-26:
> you cannot use the trackpoint-enabled firmware with Windows currently, due to incompatibilities with coolstar's Windows drivers

Can You guide me how can i reflash the right firmware again and from which Operating system i can do that .

Thankyou

MrChromebox, 2023-12-26:
there's no guide, you boot Linux, run the script, and when prompted choose the TP enabled version