MrChromebox/scripts #341: SCHCI REGISTER DUMP, Lenovo Yogo11e Intel Braswell ULTIMA

HighOnVoltage, 2023-12-27:
Hi mad lads,

Although I've managed to install coreboot via MrChromebox script, doesn't matter the Linux OS I always seem to have this issue upon installation. Tried Debian, Manjaro, Arch & Ubuntu with all the same problems upon installing data to the MMC drive. The live distros work and I can even use gparted on the MMC drive to wipe without fail. 

My EFI 2 boot partitions are only 4mb each and my understanding reading the Arch wiki I most likely need a 512MB FAT EFI boot partition. Done heaps of research of others with a similar issue and nothing was consultive. Even updated to the lastest EFI rom from MrChromebox! 

I've also tried several different brands of SD & USB drives to install media thinking one was corrupt, yet this isn't the case. The result is always the when looking at my dmesg output. 

Unsure how I've screwed this up, but would be delighted in any input. 
Many thanks,

-- Trev

Could a wise dev please help a newbie chromebook linux converter out? EFI is still very new to me.
[image]
Only have screengrabs due to the nature of issue.

HighOnVoltage, 2023-12-27:
I have also tried to clear UEFI NVRAM but it comes up with an error.

MrChromebox, 2023-12-28:
this seems to be an issue with the UEFI Full ROM firmware, and I think there might be a kernel command line workaround you can use, but don't have it handy. Might need to dump the registers when booted from stock firmware + RW_LEGACY to see what's different

MArabie, 2024-01-05:
I think this is related to https://github.com/MrChromebox/firmware/issues/310#issue-1064815093

MrChromebox, 2024-01-05:
yes, appears to be the same issue of the kernel enabling command queuing when the hardware doesn't support it / is broken for some reason.