MrChromebox/scripts #342: No touchscreen on Asus Flip with Apollo Lake.

dirkmittler, 2024-01-11:
I'm sorry if I did not provide enough information right off the bat. Feel free to ask for more device info. I have a Chromebook, which I recently gave full control to Linux over, using your UEFI Full ROM Firmware. My Linux distro is Debian 12, running Plasma and Wayland. The kernel version is 6.1.0.

Normally, under Wayland, Plasma has a tab which is supposed to allow Touchscreen configuration. Under X11, it doesn't even have such a tab. But even under Wayland, when I click on that tab, it shows me "No touchscreens found."

I just updated your UEFI Firmware, without any real problems, except for the fact that the touchscreen is still not detected.

I have looked in the 'dmesg' log, and this is the closest I was able to find:

```
[    5.191515] elan_i2c i2c-ELAN0000:00: supply vcc not found, using dummy regulator
[    5.214946] ACPI: AC: AC Adapter [AC] (off-line)
[    5.223314] cr50_i2c i2c-GOOG0005:00: cr50 TPM 2.0 (i2c 0x50 irq 57 id 0x28)
[    5.270761] cros-usbpd-notify-acpi GOOG0003:00: Couldn't get Chrome EC device pointer.
[    5.292846] Adding 999420k swap on /dev/mmcblk1p3.  Priority:-2 extents:1 across:999420k SSFS
[    5.305967] elan_i2c i2c-ELAN0000:00: Elan Touchpad: Module ID: 0x0061, Firmware: 0x0006, Sample: 0x000c, IAP: 0x000c
[    5.325083] input: Elan Touchpad as /devices/pci0000:00/0000:00:17.0/i2c_designware.4/i2c-4/i2c-ELAN0000:00/input/input5
```

In other words, the touchpad works 100%, but the touchscreen no. Because the dmesg log is mainly clean, it looks to me like there is not even the device being detected, for the kernel to go looking for the correct driver.

It's not extremely important, but would be nice to get the touchscreen working.

Sincerely,
Dirk

MrChromebox, 2024-01-11:
please provide a cbmem boot log so we can verify the firmware is detecting/initializing the touchscreen

dirkmittler, 2024-01-11:
Could you please let me know, what command(s) I must type in, to obtain a "cbmem boot log"? I have never done this before.

Dirk

dirkmittler, 2024-01-11:
Sorry to bump. But I think I've made some progress with this (since the output above seems to suggest, that Mr. Chromebox is recognizing the ELAN Touchscreen as well as the ELAN Touchpad). If I run the following command...

```
apt-file search elan.ko
```

I get copious output, telling me that the kernel module has been compiled into the kernel by devs. However, the following command produces no output at all:

```
apt-file search elan_ts.ko
```

Or, for that matter,

```
apt-file search elan_ts
```

What this would seem to suggest is, that Debian devs have compiled to module for the touchpad into the kernel, but have not compiled the module for the touchscreen. And if this interpretation is correct, the problem is not a responsibility of Mr. Chromebox. Would you agree?

Dirk

MrChromebox, 2024-01-11:
the firmware is definitely not the issue here I'd say. You can always live boot a Ubuntu or PopOS USB and the touchscreen should work OOTB there

dirkmittler, 2024-01-11:
One thing which I have done now was, to install the package 'xserver-xorg-input-synaptics', and to boot into an X11 session. But, the only effect this seemed to have was, that I get to see the fancier settings panel, for my actual touch-pad (which I've seen many times before). So because, as I had written at the beginning of this thread, touch-screen support is not critical for me, I will just have to accept, that this is one thing which will not work for me, under Debian.

Anyhow, thank you for your time.

Dirk

closed 2024-01-11 by dirkmittler

MrChromebox, 2024-01-12:
> I will just have to accept, that this is one thing which will not work for me, under Debian.

I don't see why this is, there's no configuration needed normally, just the elan_ts module loaded

dirkmittler, 2024-01-12:
First off, during a previous posting, I had made some mistakes. I had posted an incorrect link, which did not serve to substantiate my case. This would be the correct link:

[https://bugzilla.kernel.org/show_bug.cgi?id=207759](url)

Secondly, there is a fact about kernel-compiling, which I had not known, until this very experience. There exist in-tree kernel modules, which the maintainers can just configure, should not be compiled with their kernel builds.

So, there was the kernel module which was really named 'elan_i2c', which my distro ships with, and which gave me the touchpad, and then there is the module 'elants_i2c', which the Debian maintainers have chosen - up to Debian 12 - not to build.

I think that one of the ways in which Debian becomes more stable than most kernel builds is, when there is a kernel module which often causes problems, they just elect not to build it. So, according to the correct link above, a problem with this module seemed to be, that it will sometimes falsely claim a device, causing that device no longer to work.

What I have now done is, to custom-compile that kernel module. The way one does that is, to download the entire kernel source, but then only to build the parts of it which one needs, as described in this link:

[https://forums.debian.net/viewtopic.php?p=788022#p788022](url)

Believe it or not, when I gave the `insmod` command, my touch-screen worked perfectly! So now, I have a Debian 12 Chromebook, with a working touchscreen!

Now, there is a way in which what I did was not proper. I did not make this a DKMS module, simply because I don't know how. If mine were a DKMS build, every time I'd get a kernel update, and the update to the header files, the module would automatically get recompiled. What I did instead was, just to unsubscribe from future kernel updates, and to write my own `.service` file, that inserts my once-built module after every boot.

This way, on a Chromebook with only 32GiB of raw storage, I was also able to delete the kernel source code, and the kernel headers, and free up a lot of storage. The uncompressed source code takes up 1.8GB of storage!

Sincerely,
Dirk