MrChromebox/scripts #345: Elan Touchpad on Linux Ubuntu / Kernel 6.5.x / Kernel 6.x / Kernel 6.7.x

jonas5, 2024-01-14:
Bios ver: 4.22.0-5

I did a bios update yesterday on my Lenovo Morphius C13.

Got some issues booting kernel into newer versions. This was working on 4.21 November release 

Got a stack trace photo from kernel output.
[image]

The laptop still boots in kernel 6.2, newer kernels have issues with the Touchpad and most other devices loaded. It won't even load any network or WiFi.

Hope this helps.

MrChromebox, 2024-01-14:
nothing changed at all between the 4.21.x and 4.22.0 release w/r/t the touch input devices, i2c, or anything like that. The ACPI tables should be almost identical. This seems like a kernel issue to me

jonas5, 2024-01-14:
Thanks for your response, This could be a kernel issue.

I installed 6.5.x and 6.6.9 / 6.7 before I did a bios update.
The issue only occurred after the bios update.
Never the less, kernel 6.2.x is working like a charm.

jonas5, 2024-01-14:
Is there any possibility to get access to older releases to verify?

Firminator, 2024-01-23:
@jonas5 I also noticed no touchpad in kernel 6.5
5.15 LTS works though. Haven't tried 6.6 or 6.2 yet.

MrChromebox, 2024-02-27:
closing since this is not a script issue. OS issues are handled by the chrultrabook project

closed 2024-02-27 by MrChromebox