MrChromebox/scripts #347: Problem with Firmware Utility Script // asus chromebook CX34

jigirniut, 2024-01-17:
[image]
Google_Marasov.14505.527.0
asus chromebook CX34

MrChromebox, 2024-01-17:
this appears to be an issue with the UFS-equipped devices

MrChromebox, 2024-02-06:
issue has been fixed, but UFS devices are having issues with the UEFI firmware so I wouldn';t recommend flashing just yet

closed 2024-02-06 by MrChromebox

MrChromebox, 2025-06-03:
a SuzyQ cable is the only way to disable WP on the CX34. Opening the device will do nothing at all.

I'd also advise you to check the chrultrabook docs and forum to see how well your device is supported for the OS you want to run. Firmware support is only half the equation. If you have further questions I'd advise asking there as this is an issue tracker, not support forum.