MrChromebox/scripts #349: I get this error, every time I try to run the script.

hansoppa, 2024-01-19:
[image]

[image]

[image]

hansoppa, 2024-01-19:
I have no idea what to do, I've been looking everywhere for a solution.

hansoppa, 2024-01-19:
I'm also really new to chromebook, all I'm trying to do, is to get windows software. I've been at it for 2 weeks now :(
Also yes, CR50. If that even means anything.

hansoppa, 2024-01-19:
I am also in developer mode with WP disabled.
[image]

Firminator, 2024-01-21:
Try pinging raw.githubusercontent.com to see if you connectivity.

NaadeemKhali, 2024-01-25:
I'm facing the same issue even though my device is supported "Grabbiter"

hansoppa, 2024-02-03:
Hello! I got past this issue somehow. I am a complete novice at this, but I am now stuck on 'select a driver to install'. Hence, I am trying to install Windows 11 on my Lenovo Gaming 16". For some reason the intel drivers I put on my USB were incompatible with my device, so I am stuck in a loop for trying to find solutions for this situation.

closed 2024-02-06 by MrChromebox