MrChromebox/scripts #357: FW Script did not ask me to backup ROM

kkilobyte, 2024-02-02:
I have an HP Chromebook 11 G6, which corresponds to "snappy / alan", and after removing the battery and running the script, it successfully installed coreboot, however it did not ask me to backup my original ROM like it did on my octopus. Considering I literally don't have the original ROM, how do I even obtain it?

MrChromebox, 2024-02-02:
>  it did not ask me to backup my original ROM like it did on my octopus

it almost certainly did, you just skipped through it. The backup prompt is conditional on the device running stock firmware.

> Considering I literally don't have the original ROM, how do I even obtain it?

if you want to go back to stock, then create a recovery USB for your device and then select the option from the script to restore from a recovery USB.

closed 2024-02-02 by MrChromebox

kkilobyte, 2024-02-02:
@MrChromebox with all due respect, it did not ask me, but I was moving from SeaBIOS rw_legacy to OpenCore UEFI.

I was asked to backup on octopus with OpenCore rw_legacy to OpenCore UEFI, but Idek.

Thanks for the tutorial on how to restore the stock ROM tho.

[image]

MrChromebox, 2024-02-03:
I have to assume that pic is several weeks old, since it shows an older firmware version being downloaded?

I don't see how the backup prompt got skipped, it doesn't depend on if you're booted via RW_LEGACY. Unless the firmware version was unable to be read.

FYI, my firmware has nothing to do with OpenCore, that's a MacOS thing IIRC.

OtterCodes101, 2024-02-03:
> I have to assume that pic is several weeks old, since it shows an older firmware version being downloaded?
>
> I don't see how the backup prompt got skipped, it doesn't depend on if you're booted via [...]

pretty sure he means coreboot

kkilobyte, 2024-02-03:
> I have to assume that pic is several weeks old, since it shows an older firmware version being downloaded?
>
> I don't see how the backup prompt got skipped, it doesn't depend on if you're booted via [...]

hi yeah i meant coreboot, i have someone who messes with macOS a lot and i keep confusing the two