MrChromebox/scripts #362: issues with script when run on custon os

Mrnermaler, 2024-02-07:
HP chromwbook 11 g5 ee (relm) dont know wahst the problem
MrChromebox Firmware Utility Script starting up

Downloading supporting files...

Downloading required tools...

Getting device/system info...

Flashrom is unable to read current firmware; cannot continue:

/tmp/flashrom: /lib/x86_64-linux-gnu/libpci.so.3: version `LIBPCI_3.5' not found (required by /tmp/flashrom)
/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/flashrom)
/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/flashrom)
/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/flashrom)

You may need to add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu 23.04+).

See https://www.flashrom.org/FAQ for more info.
MrChromebox Firmware Utility Script starting up

Downloading supporting files...

Downloading required tools...

Getting device/system info...

Flashrom is unable to read current firmware; cannot continue:

/tmp/flashrom: /lib/x86_64-linux-gnu/libpci.so.3: version `LIBPCI_3.5' not found (required by /tmp/flashrom)
/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/flashrom)
/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/flashrom)
/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/flashrom)

You may need to add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu 23.04+).

See https://www.flashrom.org/FAQ for more info.

MrChromebox, 2024-02-07:
> HP chromwbook 11 g5 ee (relm) dont know wahst the problem

as per the errors, flashrom cannot be run because your system doesn't have libpci 3.5 or later. You could have simply googled `LIBPCI_3.5' not found` and gotten the answer. The easiest solution, assuming your system is up to date, is to just live boot a newer distro like Ubuntu 24.10 and run the script from there

Mrnermaler, 2024-02-11:
uhh ubuntu 24.04 is not out yet

ferrellsl, 2024-02-11:
> uhh ubuntu 24.04 is not out yet

Uhh, you can get 24.04 here:  https://cdimage.ubuntu.com/daily-live/current/

Mrnermaler, 2024-02-11:
ubuntu 24.04 didn't work so I used Ubuntu 23.04 it worked

closed 2024-02-11 by MrChromebox