MrChromebox/scripts #365: ERROR message 'iomem=relaxed'  ASUS CN60 chromebox

StarMusa88, 2024-02-13:
Thanks for your amazing work ...Everything is working no issues since I installed MrChromebox on my ASUS CN60 chromebox. But I also recently installed Ubuntu 23.04 and i ran the script from the firmware guide on MrChromebox.tech website.
 ( cd; curl -LOk mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh) but I keep getting error  message when trying to run the script ..

And i keep getting an ERROR Message:

You may need to add 'iomem=relaxed'  to your kernel parameters or trying running from a live USB with a more permissive kernel (eg, Ubuntu 23.04+)

Can't seem to figure out how to fix it.  Any help would be greatly appreciated.

MrChromebox, 2024-02-13:
you should have no issues running the script on Ubuntu 23.04, it works fine on a live USB here. If the installed version behaves differently, you might need to do what the error says -- add 'iomem=relaxed' to your kernel boot parameters.

PS - you don't need the 'k' in the script command, not sure why you're using it.

StarMusa88, 2024-02-13:
Hi,

 I'm still getting error message after adding 'iomem=relaxed' to kernel parameters,  I also upgraded to Ubuntu  23.10. 

lsb_release -a  15:50:17 
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 23.10
Release:	23.10
Codename:	mantic

cat /proc/cmdline
BOOT_IMAGE=/boot/vmlinuz-6.5.0-17-generic root=UUID=4a46ba30-61de-45cb-8ab1-13669fffa628 ro iomem=relaxed quiet splash vt.handoff=7

cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh

MrChromebox Firmware Utility Script starting up

Downloading supporting files...

Downloading required tools...

Getting device/system info...

Flashrom is unable to read current firmware; cannot continue:

flashrom 1.4.0-devel on Linux 6.5.0-17-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
ERROR: Could not get I/O privileges (Operation not permitted).
Your kernel may prevent access based on security policies.
Issue a 'dmesg | grep flashrom' for further information
Error: Programmer initialization failed.

You may need to add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu 23.04+).

See https://www.flashrom.org/FAQ for more info.

MrChromebox, 2024-02-13:
please try with a 2304 live USB and let me know if you have an issue there

StarMusa88, 2024-02-14:
I just tried it again on 23.04 USB live same error message..
all good it was worth a try. thanks again

MrChromebox, 2024-02-14:
this doesn't make any sense. I just booted Ubuntu 23.04 Live USB, installed curl, and ran the script without issue on my CN60

[image]

EpicPotatohead, 2024-02-16:
This started happening to me a week or two (or three...) ago on a GALLOP chromebook running Fedora 39. Perhaps something has changed with the kernel? If I reboot and add 'iomem=relaxed' to the kernel options in grub, firmware-util.sh runs normally. If I don't add that option, I get this:

```MrChromebox Firmware Utility Script starting up

Downloading supporting files...

Downloading required tools...

Getting device/system info...

Flashrom is unable to read current firmware; cannot continue:

flashrom 1.4.0-devel on Linux 6.7.4-200.fc39.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Cannot open file stream for /dev/mtd0
coreboot table found at 0x76b58000.
Found chipset "Intel Jasper Lake".
Enabling flash write... Warning: Setting BIOS Control at 0xdc from 0x8b to 0x89 failed.
New value is 0x8b.
Error accessing SPIBAR, 0x1000 bytes at 0x00000000fe010000
/dev/mem mmap failed: Operation not permitted
FAILED!
FATAL ERROR!
Error: Programmer initialization failed.

You may need to add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu 23.04+).

See https://www.flashrom.org/FAQ for more info.
```

Fantastic work, btw. I've been running Fedora on this thing for many months without issue. It's much more useful to me than ChromeOS and it works great. This is the first problem I've run into (sound doesn't work, but I don't really care), and so far the iomem=relaxed workaround works.

StarMusa88, 2024-02-16:
I'm using dual boot could that be what's causing this issue?

StarMusa88, 2024-02-16:
[image]

[image]

[image]

StarMusa88, 2024-02-16:
And this time with adding iomem=relaxed. 

[image]

[image]

StarMusa88, 2024-02-16:
Is there another way to update the bios?

[image]

MrChromebox, 2024-02-16:
@EpicPotatohead please see the chrultrabook docs for info on getting audio to work under Linux

@StarMusa88 seems like something must have changed in the default kernel config. Try using 22.04 instead? you'll likely need to install curl via apt first.

StarMusa88, 2024-03-12:
Hi @MrChromebox - I have an update, I finally figured what the issue was . I had secure boot enabled that's what was preventing the script from executing.  I was able to update the bios with no error message...Thanks again for your help much appreciated.

closed 2024-03-12 by StarMusa88

MrChromebox, 2024-03-12:
@StarMusa88 thanks for circling back :)