MrChromebox/scripts #368: i keep getting this error when trying flash firmware does any one know how to fix it

Mrnermaler, 2024-02-17:
ChromeOS Device Firmware Utility Script [2024-01-07] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**   Device: Samsung Chromebook 4 (BLUEBIRD)
** Platform: Intel GeminiLake
**  Fw Type: Stock ChromeOS
**   Fw Ver: Google_Relm.7287.370.0 (08/06/2017)
**    Fw WP: Disabled
*********************************************************
**      1) Install/Update RW_LEGACY Firmware 
** [WP] 2) Install/Update UEFI (Full ROM) Firmware 
** [WP] 3) Set Boot Options (GBB flags) 
** [WP] 4) Set Hardware ID (HWID) 
*********************************************************
Select a numeric menu option or
R to reboot  P to poweroff  Q to quit 
2

Install/Update UEFI Full ROM Firmware

IMPORTANT: flashing the firmware has the potential to brick your device, 
requiring relatively inexpensive hardware and some technical knowledge to 
recover.Not all boards can be tested prior to release, and even then slight 
differences in hardware can lead to unforseen failures.
If you don't have the ability to recover from a bad flash, you're taking a risk.

You have been warned.

Also, flashing Full ROM firmware will remove your ability to run ChromeOS.

Do you wish to continue? [y/N] y

NOTICE: flashing your Chromebook is serious business. 
To ensure recovery in case something goes wrong when flashing,
be sure to set the ccd capability 'FlashAP Always' using your 
USB-C debug cable, otherwise recovery will involve disassembling
your device (which is very difficult in some cases).

If you wish to continue, type: 'I ACCEPT' and press enter.

I ACCEPT

NOTE: After flashing UEFI firmware, you will need to install a UEFI-compatible
OS; ChromeOS will no longer be bootable. See https://mrchromebox.tech/#faq

Press Y to continue or any other key to abort. y

Connect the USB/SD device to store the firmware backup and press [Enter]
to continue.  This is non-destructive, but it is best to ensure no other
USB/SD devices are connected. 

Devices available:

1) ANKEJE USB3.0 (465.76 GB)

Enter the number for the device to be used for firmware backup: 1

Saving firmware backup as stock-firmware-BLUEBIRD-20240216.rom

Firmware backup complete. Remove the USB stick and press [Enter] to continue.

Downloading Full ROM firmware
(coreboot_edk2-bluebird-mrchromebox_20240120.rom)

Persisting device HWID

VPD extracted from current firmware

Disabling software write-protect and clearing the WP range

[... 612 lines cut ...]
Probing for Winbond W39V080FA, 1024 kB: probe_jedec_common: id1 0x4c, id2 0x41, id1 is normal flash content, id2 is normal flash content
Probing for Winbond W39V080FA (dual mode), 512 kB: probe_jedec_common: id1 0x11, id2 0x40, id1 parity violation, id1 is normal flash content, id2 is 
normal flash content
Probing for Winbond W49V002FA, 256 kB: probe_jedec_common: id1 0xff, id2 0xff, id1 parity violation, id1 is normal flash content, id2 is normal flash
 content
Found Winbond flash chip "W25Q64.W" (8192 kB, SPI).
This chip may contain one-time programmable memory. flashrom cannot read
and may never be able to write it, hence it may not be able to completely
clone the contents of this chip (see man page for details).
Some block protection in effect, disabling... disabled.
Reading ich descriptor... done.
Assuming chipset 'Baytrail'.
Added layout entry 00000000 - 00000fff named fd
Added layout entry 00200000 - 007fffff named bios
Added layout entry 00001000 - 001fffff named me
restore_power_management: Re-enabling power management.
Using region: "bios".

...skipping 1 line
FAILED
Restoring MMIO space at 0x7dbfe9a9a0fc
restore_power_management: Re-enabling power management.
An error occurred flashing the Full ROM firmware. DO NOT REBOOT!

Press [Enter] to return to the main menu.

MrChromebox, 2024-02-17:
well, you changed the HWID on your device from RELM to BLUEBIRD, so the script is providing you the firmware for the board it thinks you have, rather than the one you actually have.

Why TF did you change the HWID to begin with? SMH

Mrnermaler, 2024-02-17:
a error happens when i try i RW_legacy 
 ChromeOS Device Firmware Utility Script [2024-01-07] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**   Device: Samsung Chromebook 4 (BLUEBIRD)
** Platform: Intel GeminiLake
**  Fw Type: Stock ChromeOS
**   Fw Ver: Google_Relm.7287.370.0 (08/06/2017)
**    Fw WP: Disabled
*********************************************************
**      1) Install/Update RW_LEGACY Firmware 
** [WP] 2) Install/Update UEFI (Full ROM) Firmware 
** [WP] 3) Set Boot Options (GBB flags) 
** [WP] 4) Set Hardware ID (HWID) 
*********************************************************
Select a numeric menu option or
R to reboot  P to poweroff  Q to quit 
1

Install/Update RW_LEGACY Firmware (AltFw / edk2)

NOTE: RW_LEGACY firmware cannot be used to run Windows. Period.
If you are looking to run Windows, see the documentation on coolstar.org.
MrChromebox does not provide any support for running Windows.

Press Y to continue or any other key to return to the main menu. y

Downloading RW_LEGACY firmware update
(rwl_altfw_glk-mrchromebox_20230110.bin)

Installing RW_LEGACY firmware

flashrom 8401ef77 on Linux 4.19.241-13212-ge9b0bab46475 (x86_64)
flashrom was built with LLVM Clang 14.0.0 (/var/tmp/portage/sys-devel/llvm-14.0_pre450784_p20220316-r16/work/llvm-14.0_pre450784_p20220316/clang 282c83c32384cb2f37030c28650fef4150a8b67c), little endian
Command line (7 args): /usr/sbin/flashrom -p internal -w -i RW_LEGACY:rwl_altfw_glk-mrchromebox_20230110.bin -o /tmp/flashrom.log
-i argument specified, set fmap.
Acquiring lock (timeout=180 sec)...
Opened file lock "/run/lock/firmware_utility_lock"
Lock acquired.
Initializing internal programmer
/sys/class/mtd/mtd0 does not exist
Found candidate at: 00000500-00000528
Found coreboot table at 0x00000500.
Found candidate at: 00000000-000003c8
Found coreboot table at 0x00000000.
coreboot table found at 0x7cc13000.
coreboot header(24) checksum: 4123 table(944) checksum: 296a entries: 21
Vendor ID: GOOGLE, part ID: Relm
Using Internal DMI decoder.
DMI string chassis-type: "Desktop"
DMI string system-manufacturer: "GOOGLE"
DMI string system-product-name: "Relm"
DMI string system-version: "1.0"
DMI string baseboard-manufacturer: ""
DMI string baseboard-product-name: ""
DMI string baseboard-version: ""
Found chipset "Intel Braswell" with PCI ID 8086:229c.
Enabling flash write... Root Complex Register Block address = 0xfed1c000
GCS = 0xc01: BIOS Interface Lock-Down: enabled, Boot BIOS Straps: 0x3 (SPI)
Top Swap: not enabled
0xfff80000/0xffb80000 FWH IDSEL: 0x0
0xfff00000/0xffb00000 FWH IDSEL: 0x0
0xffe80000/0xffa80000 FWH IDSEL: 0x1
0xffe00000/0xffa00000 FWH IDSEL: 0x1
0xffd80000/0xff980000 FWH IDSEL: 0x2
0xffd00000/0xff900000 FWH IDSEL: 0x2
0xffc80000/0xff880000 FWH IDSEL: 0x3
0xffc00000/0xff800000 FWH IDSEL: 0x3
0xfff80000/0xffb80000 FWH decode enabled
0xfff00000/0xffb00000 FWH decode enabled
[... 600 lines cut ...]
Added layout entry 003e0000 - 003effff named RW_MRC_CACHE
Added layout entry 003f0000 - 003f3fff named RW_ELOG
Added layout entry 003f4000 - 003f7fff named RW_SHARED
Added layout entry 003f4000 - 003f5fff named SHARED_DATA
Added layout entry 003f6000 - 003f7fff named VBLOCK_DEV
Added layout entry 003f8000 - 003f9fff named RW_VPD
Added layout entry 003fa000 - 003fffff named RW_UNUSED
Added layout entry 00400000 - 005fffff named RW_LEGACY
Added layout entry 00600000 - 007fffff named WP_RO
Added layout entry 00600000 - 00603fff named RO_VPD
Added layout entry 00604000 - 0060ffff named RO_UNUSED
Added layout entry 00610000 - 007fffff named RO_SECTION
Added layout entry 00610000 - 006107ff named FMAP
Added layout entry 00610800 - 0061083f named RO_FRID
Added layout entry 00610840 - 00610fff named RO_FRID_PAD
Added layout entry 00611000 - 006fffff named GBB
Added layout entry 00700000 - 007fffff named BOOT_STUB
Using region: "RW_LEGACY":"rwl_altfw_glk-mrchromebox_20230110.bin".
Error: Image size (1835008 B) doesn't match the expected size (2097152 B)!
FAILED
Restoring MMIO space at 0x7af9e1f540fc
restore_power_management: Re-enabling power management.
An error occurred flashing the RW_LEGACY firmware.

Press [Enter] to return to the main menu.

Mrnermaler, 2024-02-17:
> well, you changed the HWID on your device from RELM to BLUEBIRD, so the script is providing you the firmware for the board it thinks you have, rather than the one you actually have.
> 
> Why TF did you change the HWID to begin with? SMH

when i got the Chromebook my HWID was malformed so maybe when i was copying the HWID from my doc i must have aaccidentally copyed the one for BLUEBIRD

Mrnermaler, 2024-02-17:
so it should work after a restart because i fixed the HWID

Mrnermaler, 2024-02-17:
So I installed it but the UEFI Firmware says the incorrect chromebook

Mrnermaler, 2024-02-17:
[image]

MrChromebox, 2024-02-17:
There is no problem here. If you look at https://mrchromebox.tech/#devices, you'll see that there are numerous possibilities for RELM. It's not possible to distinguish between them. So I just picked the most popular one to use.

closed 2024-02-17 by MrChromebox