MrChromebox/scripts #369: mrchromebox.tech is down

lukeb2e, 2024-02-22:
Hello,

the linked website in the readme (https://mrchromebox.tech/) is currently unavailable.

MrChromebox, 2024-02-22:
working on it

rosanyadav08, 2024-02-22:
How much time it will take to be done

MrChromebox, 2024-02-22:
no idea

rosanyadav08, 2024-02-22:
Can we get any other way to process it

MrChromebox, 2024-02-22:
don't you think I would have provided that info if it were an option?

Mrnermaler, 2024-02-22:
you can get the firmware script to download by running sudo bash -c "cd; curl -LOk https://raw.githubusercontent.com/MrChromebox/scripts/master/firmware-util.sh && sudo bash firmware-util.sh"
into ternmal aussming your not installing firmware

Mrnermaler, 2024-02-22:
you can restore stock chrome os you can change gbb flags or even change HWID

Mrnermaler, 2024-02-22:
@MrChromebox do you know whats causing your site to not work like a ddos or a bug in whats ever you use to host like windows server or apache

Mrnermaler, 2024-02-22:
you can still ping mrchromebox.tech with crosh PING 198.100.145.128 (198.100.145.128) 56(84) bytes of data.

this is mrchromebox website ip

MrChromebox, 2024-02-22:
JFC people, have some damn patience.

Mrnermaler, 2024-02-22:
i am im just saying and wondering what cause the website to stop working

MrChromebox, 2024-02-22:
this thread strongly suggests otherwise

guicompeng, 2024-02-23:
[image]

Emperor-24, 2024-02-23:
@MrChromebox I appreciate the work you are doing here, great. But please I have clients waiting to take their chromebook I wanted to download the Uefi firmware and run windows on the chromebook, the clients are not being patient. Is there any other way I can do that?

rosanyadav08, 2024-02-23:
Sir we are waiting for your site to be restored, 😞

Emperor-24, 2024-02-23:
This is really frustrating and hindering loads of jobs and profits too.. A feedback from @MrChromebox would be great

ferrellsl, 2024-02-23:
> This is really frustrating and hindering loads of jobs and profits too.. A feedback from @MrChromebox would be great

@rosanyadav08, @Emperor-24, @Mrnermaler 

Do you guys realize that Mr. Chromebox does this work for free and that he has a real job elsewhere to pay his bills?  You guys are the most impatient pricks that I have ever seen.  Since you're using his work to make profits maybe you should consider sending him a donation instead of whining about his site being down.

AverageJuliet, 2024-02-23:
Look if you morons really want your 20$ Chromebooks running Coreboot then compile and flash it yourself, its not hard.
Leave this man alone and let him recover his webserver in peace for fucks sake.

MrChromebox, 2024-02-23:
> This is really frustrating and hindering loads of jobs and profits too.

sounds like I really need to start charging for my firmware

MrChromebox, 2024-02-23:
easy solution -- I would have the builds compiled against the device serial, and fail to boot (or show an error) if there was a serial mismatch

ferrellsl, 2024-02-23:
> > This is really frustrating and hindering loads of jobs and profits too.
> 
> sounds like I really need to start charging for my firmware

Maybe add a line of text to the boot splash screen that says if you paid for the firmware that you should ask for a refund.

WeirdTreeThing, 2024-02-23:
Brick the chromebook if there is a serial mismatch, then offer a paid unbricking service.

rosanyadav08, 2024-02-23:
> > This is really frustrating and hindering loads of jobs and profits too.
> 
> sounds like I really need to start charging for my firmware

Sorry sir but we really respect you and the work you are doing it helps everyone one

AverageJuliet, 2024-02-23:
[image]

Mrchromebox  i want a refund....

wearrrrr, 2024-02-23:
> sounds like I really need to start charging for my firmware

Honestly at this point maybe you should, your website goes down for a day and immediately a wave of people come into github issues, for anyone who is really fucking impatient, maybe try archive.org? harassing this guy isn't gonna get you anywhere

ferrellsl, 2024-02-23:
@Emperor-24

I'm just going to reiterate what ninelore said earlier. "You shouldn't rely on one-man voluntary work for "jobs and profits" and then have the audacity to complain"

You already knew that this project is hosted on Github, so you would have already known that the work done here is for free and opensource, so stop trying to sound like you just stumbled onto the project like some sort of little lost lamb, and stop trying to portray yourself as some sort of victim.  All you've done is complain since your very first post.

Emperor-24, 2024-02-23:
@MrChromebox Thanks for your polite response. Is there anyway you'll need an assistance from us in getting your server back up?

ferrellsl, 2024-02-23:
> @MrChromebox Thanks for your polite response. Is there anyway you'll need an assistance from us in getting your server back up?

Maybe you should funnel some of your profits to Mr. Chromebox so he has an incentive to get the site back up for you.

Emperor-24, 2024-02-23:
@ferrellsl
 Must you people always sound so rude and insulting? Is this how you are? Are you a bullie?

Am just getting to know about Github some hours ago and signed up few hours ago too don't you get that.. I recently learnt how to run windows on chromebook some days back and the @MrChromebox site went down..

Why are you always feeling everyone is lying.

ferrellsl, 2024-02-23:
> @ferrellsl Must you people always sound so rude and insulting? Is this how you are? Are you a bullie?
>
> Am just getting to know about Github some hours ago and signed up few hours ago too don't you [...]

Must you always complain?  Must you always be a  "victim"?  I don't feel that everyone is a liar, but I do feel that YOU are a liar. 
Why are you lying about not knowing that this is a free and opensource project hosted on github?  The entire purpose of github is to host  free and opensource projects, so you obviously knew that, yet you complained anyway about losing profits and customers when Mr. Chromebox's script site went down.  And why won't you even address the fact that Mr. Chromebox gladly accepts donations for his work and then not send him a donation since you're obviously making money from his hard work?  I sent him $50 USD some months ago for his fantastic work.  Why do you think that it's OK to make a profit from free and opensource work and then complain about it when you're losing money and then come back again to play the victim?  At this point I hope the site stay down just to spite you and that when/if it does come back online, that Mr. Chromebox starts charging a fee to keep pricks like you away.

OtterCodes101, 2024-02-23:
if you guys REALLY need the firmware, just compile it yourself 😒

closed 2024-02-24 by MrChromebox