MrChromebox/scripts #373: Firmware utility fails to update firmware, but shows no error

CStafford-14, 2024-02-25:
Hi! I have an Acer R11 C738T (Cyan) that has been running well for a while. This morning, I updated the firmware using the utility and my touchscreen stopped working. It also booted immediately and when I ran the utility again, it offered to install the same update again, indicating it didn't actually get installed.

I ran the tool again, it "installed successfully," and rebooted. Again, it booted immediately, only this time I didn't have touchscreen or trackpad support! (During this time my OS didn't detect the keyboard, so I had to use a wired mouse for this. The keyboard works fine, though, since it worked in the BIOS.)

I did it a third time and it did the same thing and still says I don't have the latest update. This could be due to an improperly versioned firmware binary, though, but seeing as nobody has opened an issue for it yet I'm unsure where to post this issue.

"Current" firmware version: `MrChromebox-4.22.2-5-g822782a1001` - (02/13/2024)
Available update: `4.23` (I think) - 02/23/2024
OS: Arch Linux
FW WP: Disabled, I think I lost the screw anyways.

TL;DR: Firmware doesn't look like it's being updated. The version doesn't change in the script and it boots normally every time.

MrChromebox, 2024-02-25:
there is no problem here. The firmware source date (2/13) is older than the build date (2/23). 4.22-05 is the latest/correct version

CStafford-14, 2024-02-25:
Understood. Thank you!

closed 2024-02-27 by MrChromebox