MrChromebox/scripts #375: CN65 restore stock firmware from shellball image failed with "Image size (267 B) doesn't match the expected size (16777216 B)"

ben-z, 2024-02-27:
I'm trying to restore stock firmware on a CN65 using the `Restore using a shellball image` option. However, it's failing with `Error: Image size (267 B) doesn't match the expected size (16777216 B)!`. Do you know how I can fix this?

Full logs:

```
 ChromeOS Device Firmware Utility Script [2024-02-23]
 (c) Mr Chromebox <mrchromebox@gmail.com>
*********************************************************
**   Device: Asus Chromebox 3 / CN65 (TEEMO)
** Platform: Intel KabyLake
**  Fw Type: Full ROM / UEFI
**   Fw Ver: MrChromebox-4.22.2-5-g822782a1001 (02/13/2024)
**           Update Available (02/23/2024)
**    Fw WP: Disabled
*********************************************************
** [WP] 1) Install/Update UEFI (Full ROM) Firmware
** [WP] 2) Restore Stock Firmware
**      C) Clear UEFI NVRAM
*********************************************************
Select a numeric menu option or
R to reboot  P to poweroff  Q to quit
2

Restore Stock Firmware

Standard disclaimer: flashing the firmware has the potential to
brick your device, requiring relatively inexpensive hardware and some
technical knowledge to recover.  You have been warned.

Do you wish to continue? [y/N] y

Please select an option below for restoring the stock firmware:

1) Restore using a shellball image (downloaded) [if available]
2) Restore using a firmware backup on USB
3) Restore using a ChromeOS Recovery USB
Q) Quit and return to main menu

? 1

That's ok, I'll download a shellball firmware for you.

Confirm system details:

Device: Asus Chromebox 3 / CN65
Board Name: TEEMO

? [y/N] y

Downloading shellball.teemo.bin

VPD extracted from current firmware

Merging VPD into shellball/recovery image firmware

Restoring stock firmware

flashrom 1.4.0-devel on Linux 5.15.0-94-generic (x86_64)
flashrom was built with GCC 11.4.0, little endian
Command line (10 args): /tmp/flashrom -p internal --ifd -i bios -N -w /tmp/stock-firmware.rom -o /tmp/flashrom.log
Acquiring lock (timeout=180 sec)...
Opened file lock "/run/lock/firmware_utility_lock"
Lock acquired.
Initializing internal programmer
/sys/class/mtd/mtd0 does not exist
Found candidate at: 00000500-00000528
Found coreboot table at 0x00000500.
Found candidate at: 00000000-000004c8
Found coreboot table at 0x00000000.
coreboot table found at 0x7aaa2000.
[... 413 lines cut ...]
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
done.
Assuming chipset '100 series Sunrise Point'.
Added layout entry 00000000 - 00000fff named fd
Added layout entry 00200000 - 00ffffff named bios
Added layout entry 00001000 - 001fffff named me
restore_power_management: Re-enabling power management.
Using region: "bios".
Error: Image size (267 B) doesn't match the expected size (16777216 B)!
FAILED
Restoring PCI config space for 00:1f:5 reg 0xdc
An error occurred restoring the stock firmware. DO NOT REBOOT!

Press [Enter] to return to the main menu.
```

diredocks, 2024-02-27:
it seems that all shellball files were not found on mrchromebox.tech host:
https://www.mrchromebox.tech/files/firmware/shellball/shellball.teemo.bin
https://www.mrchromebox.tech/files/firmware/shellball/shellball.atlas.bin
....
etc

MrChromebox, 2024-02-27:
I need to update the script to remove that option; restoring from firmware backup on USB or recovery USB only

MrChromebox, 2024-02-27:
actually I found the old shellballs repo and uploaded them, so the feature should work again for supported devices

closed 2024-02-27 by MrChromebox