MrChromebox/scripts #377: Firmware Update Utility shows update, flashes it successfully, but firmware is still old

abhi-deshpande, 2024-03-08:
**Issue:**

I wanted to check updates for the firmware of my IdeaPad 3 (Octopus Lick) Chromebook, so used the firmware-util.sh as instructed on the website. Utility loads up, shows update to my firmware. When opted for update, it shows that the flash was successful. But, when a reboot is performed, the firmware date is still shown as old in system info utilities, Also, the script itself shows the update again when run again.

Tried to run it from tty as well.

**System Info:**
```
System:
  Host: octopus-lick Kernel: 6.6.14-200.fc39.x86_64 arch: x86_64 bits: 64
  Desktop: GNOME v: 45.4 Distro: Fedora Linux 39 (Workstation Edition)
Machine:
  Type: Laptop System: Google product: Lick v: rev3
    serial: <superuser required>
  Mobo: Google model: Lick v: rev3 serial: <superuser required>
    UEFI: coreboot v: MrChromebox-4.22.2-5-g822782a1001 date: 02/13/2024
Battery:
  ID-1: BAT0 charge: 36.0 Wh (93.8%) condition: 38.4/41.8 Wh (91.8%)
CPU:
  Info: dual core Intel Celeron N4020 [MCP] speed (MHz): avg: 796
    min/max: 800/2800
Graphics:
  Device-1: Intel GeminiLake [UHD Graphics 600] driver: i915 v: kernel
  Device-2: Bison EasyCamera driver: uvcvideo type: USB
  Display: wayland server: X.Org v: 23.2.4 with: Xwayland v: 23.2.4
    compositor: gnome-shell driver: X: loaded: modesetting unloaded: fbdev,vesa
    dri: iris gpu: i915 resolution: 1366x768~60Hz
  API: OpenGL v: 4.6 vendor: intel mesa v: 23.3.6 renderer: Mesa Intel UHD
    Graphics 600 (GLK 2)
Network:
  Device-1: Intel Gemini Lake PCH CNVi WiFi driver: iwlwifi
Drives:
  Local Storage: total: 58.24 GiB used: 10.88 GiB (18.7%)
Info:
  Memory: total: 4 GiB available: 3.69 GiB used: 2.46 GiB (66.7%)
  Processes: 240 Uptime: 13m Shell: Bash inxi: 3.3.33
```

**Script Output:**
```

 ChromeOS Device Firmware Utility Script [2024-02-29] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**   Device: Lenovo Ideapad 3 Chromebook (LICK)
** Platform: Intel GeminiLake
**  Fw Type: Full ROM / UEFI
**   Fw Ver: MrChromebox-4.22.2-5-g822782a1001 (02/13/2024)
**           Update Available (02/23/2024)
**    Fw WP: Disabled
*********************************************************
** [WP] 1) Install/Update UEFI (Full ROM) Firmware 
** [WP] 2) Restore Stock Firmware 
**      C) Clear UEFI NVRAM 
*********************************************************
Select a numeric menu option or
R to reboot  P to poweroff  Q to quit 
1

Install/Update UEFI Full ROM Firmware

IMPORTANT: flashing the firmware has the potential to brick your device, 
requiring relatively inexpensive hardware and some technical knowledge to 
recover.Not all boards can be tested prior to release, and even then slight 
differences in hardware can lead to unforseen failures.
If you don't have the ability to recover from a bad flash, you're taking a risk.

You have been warned.

Do you wish to continue? [y/N] y

Downloading Full ROM firmware
(coreboot_edk2-lick-mrchromebox_20240223.rom)

Persisting device HWID

VPD extracted from current firmware

Disabling software write-protect and clearing the WP range

Installing Full ROM firmware (may take up to 90s)

Full ROM firmware successfully installed/updated.

IMPORTANT:
The first boot after flashing may take substantially
longer than subsequent boots -- up to 30s or more.
Be patient and eventually your device will boot :)

Press [Enter] to return to the main menu.
```
**Screenshot after re-running the script:**
[image]

abhi-deshpande, 2024-03-08:
Ohh. So my firmware is actually fully updated. But isn't it a bit misleading ? How can I know if I need to update ? Does repository commit dates have any relation ?

MrChromebox, 2024-03-08:
sigh. this is a one-time issue because I had to rebuild all of the firmware images when the site moved to a new host. Normally the last commit date and build dates will align.

abhi-deshpande, 2024-03-08:
Okay. Now I get it. Thank you for this awesome project..!

ChocolateLoverRaj, 2024-08-15:
So it is a 2 time issue now?

MrChromebox, 2024-08-15:
> So it is a 2 time issue now?

is the explanation above as to why this can happen unclear?

ChocolateLoverRaj, 2024-08-15:
I understand what causes this issue, but idk why this issue isn't considered a bug with the firmware util script. I think it is a bug worth fixing since it could happen again and make people reflash firmware unnecessarilly and confuse people.

MrChromebox, 2024-08-15:
it's not a bug with the script, it's a difference between the filename date and the internal build date. the script has no idea about the internal build date of the files on the server, only the filename date