MrChromebox/scripts #383: can't update to latest Update Available (02/23/2024)

bam80, 2024-03-24:
I updated my firmware to supposedly latest version (02/13/2024), but the script still suggests the Update is Available (02/23/2024).
If I try to update again, the firmware apparently updates but actually the version doesn't change even after reboot (and I'm not sure it even actually flashes this time as the time spent is too small),
so the whole process can be repeated again:
```
 ChromeOS Device Firmware Utility Script [2024-03-21] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**   Device: Asus Chromebox 3 / CN65 (TEEMO)
** Platform: Intel KabyLake
**  Fw Type: Full ROM / UEFI
**   Fw Ver: MrChromebox-4.22.2-5-g822782a1001 (02/13/2024)
**           Update Available (02/23/2024)
**    Fw WP: Disabled
*********************************************************
** [WP] 1) Install/Update UEFI (Full ROM) Firmware 
** [WP] 2) Restore Stock Firmware 
**      C) Clear UEFI NVRAM 
*********************************************************
Select a numeric menu option or
R to reboot  P to poweroff  Q to quit 
1

Install/Update UEFI Full ROM Firmware

IMPORTANT: flashing the firmware has the potential to brick your device, 
requiring relatively inexpensive hardware and some technical knowledge to 
recover.Not all boards can be tested prior to release, and even then slight 
differences in hardware can lead to unforseen failures.
If you don't have the ability to recover from a bad flash, you're taking a risk.

You have been warned.

Do you wish to continue? [y/N] y

Downloading Full ROM firmware
(coreboot_edk2-fizz-mrchromebox_20240223.rom)

VPD extracted from current firmware

Disabling software write-protect and clearing the WP range

Installing Full ROM firmware (may take up to 90s)

Full ROM firmware successfully installed/updated.

IMPORTANT:
The first boot after flashing may take substantially
longer than subsequent boots -- up to 30s or more.
Be patient and eventually your device will boot :)

Press [Enter] to return to the main menu.
```

MrChromebox, 2024-03-24:
it's a bug in the script currently, because normally the file date and firmware date are the same, but currently they are not

closed 2024-03-24 by MrChromebox

bam80, 2024-03-24:
Ok, I see you closed the issue - was the bug fixed?
Usually the fix is seen directly from the issue itself.

MrChromebox, 2024-03-25:
the script shows the date of the current firmware as reported by the SMBIOS tables,  which is the date of the last change in the build, not the build date. The script only knows the date embedded in the filename of the firmware file, which is the build date. Normally when I release updates, those two dates are the same, but this time they were not do to site hosting issues and having to rebuild all the images. There's no way for me to fix this short of renaming all the files to match the last change date (2/13) or releasing a new update with matching dates