MrChromebox/scripts #384: Lenova Thinkpad 11e write protect shows enabled even after screw removal

saltyok, 2024-03-27:
I am trying to change from ChromeOS to ChromeOSFlex on a Lenova Thinkpad 11e. I removed my write protect screw according to the instructions and diagram on page 22-23 in the 11e hardware manual (https://download.lenovo.com/ibmdl/pub/pc/pccbbs/mobiles_pdf/11e_chromebook_yoga_11e_chromebook_hmm_en_sp40a26992.pdf) - but when I get to the ChromeOS Device Firmware Utility Script menu for option 2 full ROM firmware- it still says that my write protect is enabled. I am doing this switch from ChromeOS to ChromeOSFlex because our elementary school's accelerated reader quizzes and I-Excel no longer work on the 11e ChromeOS version. Any ideas of what I am doing wrong?? Do I need to put a piece of electrical tape under the board on that write protect screw area so that it doesn't make contact?

MrChromebox, 2024-03-27:
1) Flex doesn't support Baytrail Chromebooks, you won't have functional audio and possibly more

2) If the script reports WP enabled, it's because ChromeOS says it's still on. If the correct screw was removed, something else is still making contact such that the WP pin is being grounded. It could be on the under side, could be on the inside of the screw hole, hard to say

MrChromebox, 2024-03-27:
they can run Linux or Windows, and they can run Flex without audio

saltyok, 2024-03-27:
what version of Windows would be best with the 11E limitations?

MrChromebox, 2024-03-27:
please see chrultrabook.com for OS stuffs, I just build the firmware

closed 2024-11-25 by MrChromebox