MrChromebox/scripts #385: coreboot-nipperkin version 20240321 can't work

hx801217, 2024-04-01:
Hello, Mrchromebook!
Here, I reflect it:
My chromebook is HP Elite C645 G2, Aka Nipperkin.
I use your script to upgraded CoreBoot from 20240223 to 20240321, but there is a problem, it can not boot Win10 or Win11, or even  WinPE. The 20240223 version can do this。 

However, the problem of 20240223, even if the audio drive of Coolstar is purchased, it cannot be correctly driven hardware to make its audio normal.
[image]

MrChromebox, 2024-04-01:
the 20240223 version was built for DEWATT, not NIPPERKIN. the 20240321 version is same code but rebuilt for the correct board. There are lots of issues with the Cezanne Chromebooks now, I wouldn't expect things to work at all TBH.

hx801217, 2024-04-01:
> 20240223 版本是为 DEWATT 而不是 NIPPERKIN 构建的。20240321 版本是相同的代码，但针对正确的板进行了重建。现在 Cezanne Chromebook 存在很多问题，老实说我根本不指望一切能正常工作。

it can't work now, I hope you can improve it as soon as possible, thank you very much.

hx801217, 2024-04-01:
> 20240223版本是为DEWATT而不是NIPPERKIN构建的。20240321版本是相同的代码，但针对正确的板进行了重建。现在Cezanne Chromebook存在很多问题，老实说我根本不指望一切都能正常工作。

Forgot to say, the performance of 0321 is that it will be directly blue screen when it starts to the Win interface. So I can only use 0223 at present.

MrChromebox, 2024-04-01:
ok, I just pushed an update for Nipperkin that should fix the BSOD issue

hx801217, 2024-04-01:
> 好的，我刚刚推送了 Nipperkin 的更新，应该可以解决 BSOD 问题

thanks lot.

hx801217, 2024-04-09:
> ok, I just pushed an update for Nipperkin that should fix the BSOD issue
I updated 0401 and looked very good, thank you.
But there are two unknown devices. it is useless, can you hide it in BIOS by next update ?

THE TWO unknown devices:
[image]
[image]

hx801217, 2024-04-09:
> ok, I just pushed an update for Nipperkin that should fix the BSOD issue
Oh, I forgot that although the CPU is actually in BOOST, the task manager does not display it. It has always displayed 2.30GHz

hx801217, 2024-04-10:
> the fingerprint reader I can hide, the sensor fusion hub has a driver that should be installed via the AMD chipset drivers.

yes,the sensor fusion hub can be drove,but no effect,only a yellow exclamation mark is displayed.the AMD chipset drivers can not drive it correct，If you have the right driver, can you provide it here。

MrChromebox, 2024-04-10:
I don't provide support for Windows, don't know about drivers etc. The SFH device cannot be disabled in firmware

MrChromebox, 2024-04-16:
closing as the firmware side of these issues is resolved

closed 2024-04-16 by MrChromebox