MrChromebox/scripts #389: Flashed Lenovo Ideapad Slim 3i Chromebook (PUJJOTEEN15W) Now Unable to Update

chipmonger1, 2024-04-02:
Flashed a Lenovo Ideapad Slim 3i Chromebook (PUJJOTEEN15W) with MrChromebox-4.22.0 (12/22/2023) (full rom/UEFI package). 
I went to update with a newer version of the firmware but am not able to do so. Running the MrChromebox script gives the following with option #1 to Install/Update UEFI greyed out.

ChromeOS Device Firmware Utility Script [2024-01-07]
(c) Mr Chromebox [mrchromebox@gmail.com](mailto:mrchromebox@gmail.com)

** Device: ERROR: unknown or unidentifiable device (PUJJOTEEN15W)
** Platform: (unrecognized)
** Fw Type: Full ROM / UEFI
** Fw Ver: MrChromebox-4.22.0 (12/22/2023)
** Fw WP: Disabled

** 1) Install/Update UEFI (Full ROM) Firmware
** [WP] 2) Restore Stock Firmware
** C) Clear UEFI NVRAM

Select a numeric menu option or
R to reboot P to poweroff Q to quit

MrChromebox, 2024-04-02:
just pushed a fix, quit and retry in a min or so

closed 2024-04-02 by MrChromebox

chipmonger1, 2024-04-02:
With the fix I was able to update the firmware.  Only issue I see is that in the boot options/settings menu that 0 MB RAM is reported (this was happening on the previous version).  Works normally otherwise.  Thank you!

MrChromebox, 2024-04-03:
could be the FSP binary needs an update, possible the next release will fix it