MrChromebox/scripts #391: Failed to Clear UEFI NVRAM

Integral-Tech, 2024-04-15:
```shell
Clearing the NVRAM will remove all EFI variables
and reset the boot order to the default.

Would you like to continue? [y/N] y

Clearing NVRAM...

flashrom 1.4.0-devel on Linux 6.8.6-hardened1-1-hardened (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Opened /dev/mtd0 successfully
Found Programmer flash chip "Opaque flash chip" (16384 kB, Programmer-specific) on internal.
Cannot read 0x000008 bytes at 0x200000: Input/output error
read_flash: failed to read (0x200000..0x200007).
Cannot read 0x000008 bytes at 0x100000: Input/output error
read_flash: failed to read (0x100000..0x100007).
Cannot read 0x000008 bytes at 0x300000: Input/output error
read_flash: failed to read (0x300000..0x300007).
Cannot read 0x000008 bytes at 0x080000: Input/output error
read_flash: failed to read (0x080000..0x080007).
Cannot read 0x000008 bytes at 0x180000: Input/output error
read_flash: failed to read (0x180000..0x180007).
Cannot read 0x000008 bytes at 0x280000: Input/output error
read_flash: failed to read (0x280000..0x280007).
Cannot read 0x000008 bytes at 0x380000: Input/output error
read_flash: failed to read (0x380000..0x380007).
Cannot read 0x000008 bytes at 0x040000: Input/output error
read_flash: failed to read (0x040000..0x040007).
Cannot read 0x000008 bytes at 0x0c0000: Input/output error
read_flash: failed to read (0x0c0000..0x0c0007).
Cannot read 0x000008 bytes at 0x140000: Input/output error
read_flash: failed to read (0x140000..0x140007).
Cannot read 0x000008 bytes at 0x1c0000: Input/output error
read_flash: failed to read (0x1c0000..0x1c0007).
Cannot read 0x000008 bytes at 0x240000: Input/output error
read_flash: failed to read (0x240000..0x240007).
Cannot read 0x000008 bytes at 0x2c0000: Input/output error
read_flash: failed to read (0x2c0000..0x2c0007).
Cannot read 0x000008 bytes at 0x340000: Input/output error
read_flash: failed to read (0x340000..0x340007).
Cannot read 0x000008 bytes at 0x3c0000: Input/output error
read_flash: failed to read (0x3c0000..0x3c0007).
Cannot read 0x000008 bytes at 0x020000: Input/output error
read_flash: failed to read (0x020000..0x020007).
Cannot read 0x000008 bytes at 0x060000: Input/output error
read_flash: failed to read (0x060000..0x060007).
Cannot read 0x000008 bytes at 0x0a0000: Input/output error
read_flash: failed to read (0x0a0000..0x0a0007).
Cannot read 0x000008 bytes at 0x0e0000: Input/output error
read_flash: failed to read (0x0e0000..0x0e0007).
Cannot read 0x000008 bytes at 0x120000: Input/output error
read_flash: failed to read (0x120000..0x120007).
Cannot read 0x000008 bytes at 0x160000: Input/output error
read_flash: failed to read (0x160000..0x160007).
Cannot read 0x000008 bytes at 0x1a0000: Input/output error
read_flash: failed to read (0x1a0000..0x1a0007).
Cannot read 0x000008 bytes at 0x1e0000: Input/output error
read_flash: failed to read (0x1e0000..0x1e0007).
Cannot read 0x000008 bytes at 0x220000: Input/output error
read_flash: failed to read (0x220000..0x220007).
Cannot read 0x000008 bytes at 0x260000: Input/output error
read_flash: failed to read (0x260000..0x260007).
Cannot read 0x000008 bytes at 0x2a0000: Input/output error
read_flash: failed to read (0x2a0000..0x2a0007).
Cannot read 0x000008 bytes at 0x2e0000: Input/output error
read_flash: failed to read (0x2e0000..0x2e0007).
Cannot read 0x000008 bytes at 0x320000: Input/output error
read_flash: failed to read (0x320000..0x320007).
[... 442 lines cut ...]
Cannot read 0x000008 bytes at 0x3d4000: Input/output error
read_flash: failed to read (0x3d4000..0x3d4007).
Cannot read 0x000008 bytes at 0x3dc000: Input/output error
read_flash: failed to read (0x3dc000..0x3dc007).
Cannot read 0x000008 bytes at 0x3e4000: Input/output error
read_flash: failed to read (0x3e4000..0x3e4007).
Cannot read 0x000008 bytes at 0x3ec000: Input/output error
read_flash: failed to read (0x3ec000..0x3ec007).
Cannot read 0x000008 bytes at 0x3f4000: Input/output error
read_flash: failed to read (0x3f4000..0x3f4007).
Cannot read 0x000008 bytes at 0x3fc000: Input/output error
read_flash: failed to read (0x3fc000..0x3fc007).
Using region: "SMMSTORE".
disable_power_management: Failed to open /run/lock/power_override/flashrom.lock for writing: No such file or directory
Reading old flash chip contents... Cannot read 0x001000 bytes at 0x001000: Input/output error
read_flash: failed to read (00000000..0xffffff).
FAILED.
Your flash chip is in an unknown state.
Get help on IRC (see https://www.flashrom.org/Contact) or mail
flashrom@flashrom.org with the subject "FAILED: <your board name>"!-------------------------------------------------------------------------------
DO NOT REBOOT OR POWEROFF!
FAILED

Failed to erase SMMSTORE firmware region; NVRAM not cleared.
```

MrChromebox, 2024-04-15:
looks like it might be a kernel issue, the flashrom binary used hasn't changed in quite some time. Are you able to update the firmware otherwise?

Integral-Tech, 2024-04-16:
> looks like it might be a kernel issue, the flashrom binary used hasn't changed in quite some time. Are you able to update the firmware otherwise?

```shell
*********************************************************
**   Device: HP Elite c1030 Chromebook / HP Chromebook x360 13c (JINLON)
** Platform: Intel CometLake
**  Fw Type: Full ROM / UEFI
**   Fw Ver: MrChromebox-4.22.2-5-g822782a1001 (02/13/2024)
**           Update Available (02/23/2024)
**    Fw WP: Disabled
*********************************************************
```

The latest 4.22.3 release hasn't been detected by the script yet.

MrChromebox, 2024-04-16:
> The latest 4.22.3 release hasn't been detected by the script yet.

there is no update for JINLON, only the AMD boards listed in the release notes on my site

MrChromebox, 2024-04-18:
btw, this has been identified as a flashrom issue, but I can't reproduce it on newer platforms like CML, only BDW and older

MrChromebox, 2024-04-18:
I just pushed a new flashrom build, so try again after a fresh reboot

closed 2024-06-23 by Integral-Tech