MrChromebox/scripts #393: Acer C720 coreboot image mismatch (Google:Peppy) vs (Acer:Peppy)

marmot7777, 2024-04-21:
I am trying to replace the firmware in my Acer C720....

Device: Acer C720 Chromebook (PEPPY)
Platform: Intel Haswell
Fw Type: Full ROM / Legacy
Fw Ver:
Fw WP: Disabled.

When I try to run the firmware utility script, I get the following error. 

This coreboot image (Google:Peppy) does not appear to be correct for the detected mainboard (Acer:Peppy). Aborting. You can override this with -p internal:boardmismatch=force.

I could try forcing it, but I don't know how to do that either.

MrChromebox, 2024-04-21:
what firmware is the device currently running?

you can manually for the flash by running the script and letting it error out as above, then exiting and running:

`sudo /tmp/flashrom -p internal:boardmismatch=force --ifd -i bios -w /tmp/coreboot_edk2-peppy-mrchromebox_20240416.rom --noverify`

marmot7777, 2024-04-21:
Fw Ver: (01/16/2015)

Forcing it appears to have failed too:

Proceeding anyway because user forced us to.
Reading old flash chip contents... FAILED.

marmot7777, 2024-04-21:
When booting, it says
Read-only firmware id: Google_Peppy.4389.78.0
Active firmware id: Google_Peppy.4389.93.0

MrChromebox, 2024-04-21:
I'm going to need the full log to see what's going on

`sudo /tmp/flashrom -p internal:boardmismatch=force --ifd -i bios -w /tmp/coreboot_edk2-peppy-mrchromebox_20240416.rom --noverify -o flashrom.log` 

and then attach flashrom.log here (please don't copy/paste it)

marmot7777, 2024-04-21:
Hi MrChromeBox, I attached the log file.

[flashrom.log](https://github.com/MrChromebox/scripts/files/15054626/flashrom.log)

MrChromebox, 2024-04-21:
I'm not sure exactly what the issue is, something with the new build of flashrom I switched to, and the fact that you're running very old stock firmware with bad DMI info.

let's try using an older build of flashrom - download, extract, and use this version instead of the one provided by the script.

[flashrom_cros_libpci37_20230206.tar.gz](https://github.com/MrChromebox/scripts/files/15055095/flashrom_cros_libpci37_20230206.tar.gz)

marmot7777, 2024-04-21:
Sorry! I'm good at figuring things out, but this is day 2 of Linux for me. I copied flashrom to my home directory. What commands do I need to run?

MrChromebox, 2024-04-22:
assuming you downloaded the flashrom tar.gz file to your home dir, run:

`tar zxf flashrom_cros_libpci37_20230206.tar.gz`
to extract it, then
`sudo ~/flashrom -p internal --ifd -i bios -w /tmp/coreboot_edk2-peppy-mrchromebox_20240416.rom --noverify -o flashrom.log`
to flash it

marmot7777, 2024-04-22:
The compressed file you sent me only had a single file in it..."flashrom" with no file extension. The command couldn't find the specified file, so I replaced the missing file with just "flashrom"

I hope this helps. 

sudo ~/flashrom -p internal --ifd -i bios -w /tmp/flashrom --noverify -o flashrom.log

[flashrom1.log](https://github.com/MrChromebox/scripts/files/15055877/flashrom1.log)

MrChromebox, 2024-04-22:
the compressed file was an updated flashrom binary, to replace the use of `/tmp/flashrom` from the script. I don't know why you changed the name of the file flashrom is trying to write, which should remain unchanged. Only the flashrom app/binary and the path to it in the command changes

marmot7777, 2024-04-22:
Alright, I followed your instructions exactly this time, and it appears to have worked. My computer boots with the new firmware. Thank you!

marmot7777, 2024-04-22:
Amateur question...I was able to install Linux Mint on my C720 without flashing the firmware. I just removed the WP screw, put in developer mode, and used a Linux Mint USB to install. Is there much benefit to flashing MrChromebox firmware if I am running Linux?

marmot7777, 2024-04-22:
I'm getting a larger SSD, and and a new motherboard that has more soldered on RAM. Will do a clean install of linux after making sure the MrChromeBox firmware is installed, just to cover my bases.

MrChromebox, 2024-04-22:
the ChromeOS firmware on your device hasn't been updated since 2014 probably. The fact that legacy boot mode works OOTB is an aberration, not the norm. Flashing my UEFI firmware gives you a current firmware with latest security fixes, microcode, TPM management, Secure Boot support, and no developer mode warning screen (and legacy boot which can be disabled via loss of battery power)

closed 2024-04-22 by MrChromebox