MrChromebox/scripts #397: ThinkCentre M70q chromeos locked

bparihar, 2024-04-26:
I have recently purchased ThinkCentre M70q with locked ChromeOS unable to unlock it by any method please help 

I'm already a user of  unlocked Chromebook windows user thanks to @MrChromebox for that support.

MrChromebox, 2024-04-26:
this isn't really an issue with my firmware or something my script can do. The best option I can offer is for you to flash the UEFI firmware image directly using an external programmer, following the instructions in my unbricking guide: https://wiki.mrchromebox.tech/Unbricking. a SuzyQ will not work here since the CCD is locked out by virtue of the device being managed

bparihar, 2024-04-26:
@MrChromebox Thanks for understanding the situation please share the UEFI Firmware for Lenovo M70Q So i can unlock the device

MrChromebox, 2024-04-26:
the unbricking guide linked above tells you how to get the current download links :)

bparihar, 2024-04-27:
Hello There @MrChromebox Lenovo M70q is not present in the list where can i find the firmware

MrChromebox, 2024-04-27:
> Hello There @MrChromebox Lenovo M70q is not present in the list where can i find the firmware

You mean the M60q, right? Google turns up nothing for a M70q Chromebox

https://mrchromebox.tech/#devices lists the name for the M60q as KINOX. The script provides a download for this board

bparihar, 2024-05-02:
@MrChromebox  i mean Lenovo M70Q its this machine 
https://www.lenovo.com/ca/en/p/desktops/thinkcentre/m-series-tiny/thinkcentre-m70q/wmd00000405?orgRef=https%253A%252F%252Fwww.google.com%252F

locked as chromeos and showing some Amazon ChromeOS Admin locked

MrChromebox, 2024-05-02:
@bparihar that's not a ChromeOS device, nothing we can do to help

closed 2024-05-02 by MrChromebox

MrChromebox, 2025-12-14:
> It's runing on Chromium 132.0.6834.214 csos-release-channel iridium test

that's an Amazon device running a fork of  ChromiumOS, it's not a ChromeOS device and nothing I can do / not supported