MrChromebox/scripts #399: taniks rom download hanging

iamangus, 2024-04-29:
when trying to flash the full uefi rom the script hangs at the download step. it just says:

Downloading Full ROM firmware
(coreboot_edk2-taniks-mrchromebox_20240416.rom)

Using the sources script I pieced together the below download link.

https://www.mrchromebox.tech/files/firmware/full_rom/coreboot_edk2-taniks-mrchromebox_20240416.rom

Its only 32MB, and I have very fast internet.

Any help would be appreciated.

MrChromebox, 2024-04-29:
I'm not seeing any issues on the server, and no problem downloading it myself. Not sure why some people are reporting issues

iamangus, 2024-04-29:
Interestingly I can curl -sLO that url from my main linux PC, but manually running that command from the ctrl+alt+f2 terminal hangs as it does in the script. I can run it verbose and it seems to get passed all the ssl handshake stuff and then just hangs waiting for the file. I will work on piping the output to a file and getting that somewhere that I can attach here. 

If there is anything else I can test please let me know.

iamangus, 2024-04-29:
`*   Trying 78.47.63.107:443...
* Connected to mrchromebox.tech (78.47.63.107) port 443 (#0)
* ALPN: offers h2,http/1.1
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [122 bytes data]
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
{ [35 bytes data]
* TLSv1.3 (IN), TLS handshake, Certificate (11):
{ [2610 bytes data]
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
{ [264 bytes data]
* TLSv1.3 (IN), TLS handshake, Finished (20):
{ [36 bytes data]
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
} [1 bytes data]
* TLSv1.3 (OUT), TLS handshake, Finished (20):
} [36 bytes data]
* SSL connection using TLSv1.3 / TLS_CHACHA20_POLY1305_SHA256
* ALPN: server accepted h2
* Server certificate:
*  subject: CN=mrchromebox.tech
*  start date: Apr 23 14:59:14 2024 GMT
*  expire date: Jul 22 14:59:13 2024 GMT
*  subjectAltName: host "mrchromebox.tech" matched cert's "mrchromebox.tech"
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
} [5 bytes data]
* using HTTP/2
* h2 [:method: GET]
* h2 [:scheme: https]
* h2 [:authority: mrchromebox.tech]
* h2 [:path: /files/firmware/full_rom/coreboot_edk2-taniks-mrchromebox_20240416.rom]
* h2 [user-agent: curl/8.1.2]
* h2 [accept: */*]
* Using Stream ID: 1 (easy handle 0x5679e7076e70)
} [5 bytes data]
> GET /files/firmware/full_rom/coreboot_edk2-taniks-mrchromebox_20240416.rom HTTP/2
> Host: mrchromebox.tech
> User-Agent: curl/8.1.2
> Accept: */*
> 
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [57 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [57 bytes data]
* old SSL session ID is stale, removing
{ [5 bytes data]
< HTTP/2 200 
< last-modified: Tue, 16 Apr 2024 14:00:36 GMT
< etag: "2000000-616372991f851"
< accept-ranges: bytes
< content-length: 33554432
< date: Mon, 29 Apr 2024 01:34:33 GMT
< server: Apache
< 
{ [16275 bytes data]`

iamangus, 2024-04-29:
running the curl in the background and watching the size of the file it looks like it might be downloading but at about 20KB/s. Will take about ~30 minutes to download at that rate but will report back.

MrChromebox, 2024-04-29:
it downloaded for me in <2 seconds. so IDK

iamangus, 2024-04-29:
on my desktop PC it also downloads very quickly. I was testing other files on the chromebook and the below test file downloads in about 3 seconds.

https://link.testfile.org/30MB

So no idea what about the rom is making it download so slow.

Going to try and run the uefi install script again and give it the 30 minutes it needs to download.

iamangus, 2024-04-29:
it took about 50 minutes but eventually proceeded and completed successfully!

If anyone else has this issue just be patient. I am not sure why it is so slow but it will eventually work.

If I can help with further isolating the cause of the issue please let me know. and thank you @MrChromebox for the help!

closed 2024-04-29 by iamangus