MrChromebox/scripts #400: unable to download ROM, download never finishes

shorterchild, 2024-05-01:
Good afternoon!

I am having trouble installing the ROM.  I begin the process but I can't get past the "Downloading full ROM firmware..." step.

I saw that someone else recently closed their issue with not being able to download a ROM, so I'm hoping this is related.

In their issue they were able to download the ROM via a direct link that you posted.  Where should the ROM be stored if I were to download one directly and use it within your script?

[image]

MrChromebox, 2024-05-01:
one cannot currently use the script to flash a locally-stored file, but it's something I'm working on

shorterchild, 2024-05-01:
Thank you for the quick reply.

Is there any reason I am not able to download?  I left the computer alone for about an hour and when I came back there was a red error message about a failed checksum.  I have started the download again to see if it is successful this time.

MrChromebox, 2024-05-01:
I have no idea, the server isn't reporting any issues and I'm not seeing them myself. Must be a geographic issue

shorterchild, 2024-05-01:
I disconnected from my home internet and connected via mobile hotspot (connecting from Illinois).  That solved the problem.

[image]

closed 2024-05-01 by shorterchild