MrChromebox/scripts #401: Error Flashing RW_LEGACY FW

aeonull, 2024-05-08:
Context: Running script from Debian install on Google Pixelbook (2017). I had modified the system ~5 years ago using the Mr. Chromebox program available back then with a ChromeOS/GalliumOS dual-boot setup. I am now trying to transition the laptop to a standalone Debian system as I don't intend to use ChromeOS. I read the documentation on the mrchromebox.tech site and the docs.chrultrabook.com site which both recommend UEFI Fw, but I don't have a Suzy-Q cable to disable WP. I booted to RW_LEGACY where I had the old SeaBIOS from my previous install, and installed Debian 12 from USB.

Knowing I was running a 5-year old setup, I wanted to use the script to update the RW_LEGACY Fw I was using. I ran the installer with `cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh`, selected 1, confirmed I was not running Windows with Y, and defaulted to N for booting from USB. From this point, the script returns the following:

```
Installing RW_LEGACY firmware

flashrom MrChromebox-1.4.0-devel_2024.04.18 (git:v1.2-1418-ge5ed0c6340) on Linux 6.6.25-chrultrabook (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0x7aabe000.
Found chipset "Intel Kaby Lake Y w/ iHDCP2.2 Prem.".
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... Warning: Setting BIOS Control at 0xdc from 0x8b to 0x89 failed.
New value is 0x8b.
SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG1: BIOS region (0x00200000-0x00ffffff) is read-write.
FREG2: Management Engine region (0x00001000-0x001fffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
PR0: Warning: 0x009d0000-0x009effff is read-only.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
OK.
Found GigaDevice flash chip "GD25B128B/GD25Q128B" (16384 kB, Programmer-specific) on internal.
===
This flash part has status UNTESTED for operations: WP
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
You can also try to follow the instructions here:
https://www.flashrom.org/contrib_howtos/how_to_mark_chip_tested.html
Thanks for your help!
Reading ich descriptor... done.
Using region: "bios".
Reading flash... done.
 

Press [Enter] to return to the main menu.
```

I am hoping that updating the FW might fix a couple issues where the screen doesn't dim despite the brightness keys seemingly being detected correctly (Brightness bar in desktop environment will update) and the touchpad being seemingly disabled for the session unless it's moved around during boot. I've also heard the utility can be used to set RW_LEGACY boot as the default from the Developer boot screen (which I'd also like to change to be less jarringly bright).

MrChromebox, 2024-05-09:
I'm guessing the updated flashrom build I'm using doesn't like the command line params and is failing, because the log shown is from the initial read at startup and is successful. I'll have to look into this.

That said, SeaBIOS is just a bootloader, it's not going to change anything other than the booting of the OS. You're still running the stock Google firmware from ~2015. If you want better firmware, you'll have to disable WP and flash the UEFI firmware. There are SuzyQ cables/adapters on ebay for $10

MrChromebox, 2024-11-25:
issue should be fixed, please re-open if it's not

closed 2024-11-25 by MrChromebox