MrChromebox/scripts #402: firmware missing

samascience, 2024-05-16:
https://www.mrchromebox.tech/files/firmware/full_rom/coreboot_edk2-coral-mrchromebox_20240416.rom

the firmware is missing for this build !!

MrChromebox, 2024-05-16:
it's not missing because no board should be using coral firmware. The script detects this and instructs you to select the correct board name for your device.

closed 2024-05-16 by MrChromebox

samascience, 2024-05-16:
any idea what firmware is the one to be used for 

2018-02-14 | Acer | Acer Chromebook 11 (C732, C732T, C732L & C732LT ) | Astronaut | coral | coral | x86_64 | 4.4 | x86_64 | ApolloLake | Chromebook |   | Yes
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

2018-02-14	Acer	[Acer Chromebook 11 (C732, C732T, C732L & C732LT )](https://www.acer.com/ac/en/US/press/2018/333129)	Astronaut	coral	coral	x86_64	4.4	x86_64	ApolloLake	Chromebook		Yes
2018-02-14 | Acer | Acer Chromebook 11 (C732, C732T, C732L & C732LT ) | Astronaut | coral | coral | x86_64 | 4.4 | x86_64 | ApolloLake | Chromebook |   | Yes
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

2018-02-14	Acer	[Acer Chromebook 11 (C732, C732T, C732L & C732LT )](https://www.acer.com/ac/en/US/press/2018/333129)	Astronaut	coral	coral	x86_64	4.4	x86_64	ApolloLake	Chromebook		Yes

MrChromebox, 2024-05-17:
did  you look on my site? https://mrchromebox.tech/#devices