MrChromebox/scripts #403: Firmware Script shows that firmware is unavailable to download even though it is manually downloadable.

tux7k, 2024-05-18:
Hi,

I'm trying to restore my firmware to it's stock version and I don't have the firmware file. I tried using the shellball option and it says it isn't available for my model (CELES). This is quite confusing because [this link](https://www.mrchromebox.tech/files/firmware/shellball/shellball.celes.bin) to directly download the CELES firmware works. Is there any reason the script might not be working?

Thanks.

MrChromebox, 2024-05-19:
returning to stock / shellball ROMs are not available for EOL devices, per the script documentation. If the file is on the server, it's only because I hadn't gotten around to removing it.

closed 2024-05-19 by MrChromebox

tux7k, 2024-06-03:
not going to reopen but since you hadn't removed it at the time, I forced the script to download it and flash it and it worked fine, just here to say that

MrChromebox, 2024-06-03:
obviously there's no technical limitation with going back to stock ChromeOS on an EOL device, it's just a terrible idea to be running an out of date OS/browser that isn't being updated anymore and won't get any more security fixes.