MrChromebox/scripts #404: Firmware Utility Script error

tyrelldotexe, 2024-05-19:
I haven't used my old Chromebook in years and wanted to dig it up and use it again. I installed GallumOS back when it was active and since it's retired, I want to revert my Chromebook to ChromeOS. I've done so multiple times in the past with the script, but now I get this error when I try to run it:

> MrChromebox Firmware Utility Script starting up
> 
> Downloading supporting files...
> 
> Downloading required tools...
> 
> Getting device/system info...
> 
> Flashrom is unable to read current firmware; cannot continue:
> 
> /tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/flashrom)
> /tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/flashrom)
> 
> You may need to add 'iomem=relaxed' to your kernel parameters,
> or trying running from a Live USB with a more permissive kernel (eg, Ubuntu 23.04+).
> 
> See https://www.flashrom.org/FAQ for more info.

I know this is probably a simple fix, but I'm not the most Linux-literate person. I tried adding "iomem=relaxed" to the Grub file and still got the blurb. I'm hoping someone here can help me.

MrChromebox, 2024-05-19:
>  I want to revert my Chromebook to ChromeOS.

note that this isn't supported for Chromebooks which are EOL

> /tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version GLIBC_2.33' not found (required by /tmp/flashrom) 

your OS is too old / doesn't have a new enough version of the glibc library which flashrom uses.

My recommendation is to live-boot a newer version of Linux, update the firmware to the latest release, then reboot and install a current stable linux distro. if it's a Haswell or Broadwell Chromebook, then ChromeOS Flex is an option as well

tyrelldotexe, 2024-05-19:
> this isn't supported for Chromebooks which are EOL

Reverting to ChromeOS? It worked for me in the past and my Chromebook was EOL by then. I have a Dell Chromebook 11 3120.

I have GalliumOS 3.1 installed. Should I reinstall it or something? As for live-booting, would it possible to run the script with that OS and have it work? I was thinking about using Linux Mint or Arch for it.

I apologize if this is all simple. Like I said, I'm not very Linux-literate; mostly from primarily using Windows 😅

MrChromebox, 2024-05-20:
I don't support reverting to stock ChromeOS on EOL devices, the script will warn against it and will not provide a shellball ROM, for instance.

live-booting a current Linux distro like Mint would allow you to run the script and update the firmware. You can then install Mint and everything should work OOTB.

tyrelldotexe, 2024-05-20:
Thank you so much! You have been a big help! Due to my Chromebook's small storage (16GB), I can't use Mint past the live-boot.

If you have any distro suggestions for the storage size, feel free to let me know. You've already been awesome, so don't feel forced to research it for me :)

MrChromebox, 2024-05-20:
I run Pop_OS! on mine, but had to do custom partitioning w/o the recovery partition. 1GB ESP (/boot/efi), rest for /. Have 4.4 GB free currently

closed 2024-05-20 by MrChromebox