MrChromebox/scripts #405: Question: Running firmware-util.sh from Linux

mlanner, 2024-05-26:
Hi,

I have an HP x2 SORAKA tablet, which I successfully ran the firmware-util script on several years ago. Since it was a bit of an experiment, I put it into legacy BIOS mode, as I wanted a way back to ChromeOS.

Recently I decided I wanted to "upgrade" it to UEFI and tried to run the latest version of the firmware-util script. I'm running this from inside my Ubuntu 24.04 installation. I tells me `Fw WP: Enabled`.

Since this is a Chromebook that's not easily disassembled, I'm trying to disable the write-protect using my SuzyQable (compatible) cable and I can see the USB devices show up. I then try to unlock/disable it per the instructions provided:

```
echo "wp false" > /dev/ttyUSB0
echo "wp false atboot" > /dev/ttyUSB0
```

... and then reboot and re-run the script. However, for whatever reason, it doesn't get into `Disabled` mode.

Can the write-protect only be done from ChromeOS?

MrChromebox, 2024-05-26:
The instructions for disabling firmware WP using the SuzyQ cable/via CR50 require that the CR50 be in an open (vs locked) state; opening can only be done via ChromeOS using gsctool, or by booting with the battery disconnected.

mlanner, 2024-05-26:
Hmm ... OK, guess I'm gonna need to return the tablet back to ChromeOS then. I'll have to read up on how to get there, as I don't think I have my original backup around. Thanks for the quick response.

MrChromebox, 2024-05-26:
you're running RW_LEGACY firmware, so all you do is perform a ChromeOS USB recovery. No need for original anything

mlanner, 2024-05-27:
Back to stock. Will try unlocking and installing UEFI. Thanks for the help!

closed 2024-05-27 by mlanner