MrChromebox/scripts #406: HP CHROMEBOOK C640 G2 NOT BEING BOOTED IN WINDOWS

rosanyadav08, 2024-06-05:
After flashing chrome os , when we try to boot it in UEFI method the storege / in cheap ssd not showing , what to do please can you guide ,
Please help me with your precious time

MrChromebox, 2024-06-05:
I don't understand the issue, can you attach a picture?

rosanyadav08, 2024-06-06:
[image]

[image]

I don't know what to do for this

rosanyadav08, 2024-06-06:
I am trying to make it in windows 11 but because of this error i am faceing issues

MrChromebox, 2024-06-06:
does your internal storage drive show up in the boot menu list?
what is the board name shown at the top of the UEFI settings screen?
can you boot a Linux USB and capture a cbmem log?

rosanyadav08, 2024-06-07:
Board name -  elemi 
It doesn't shows and storage drives in boot  list

MrChromebox, 2024-06-07:
sounds like an eMMC driver issue in edk2, unless something changed on the coreboot side. Do you have a NVMe drive you can pop in to test?

rosanyadav08, 2024-06-07:
The laptop doesn't have any slot in board for ssd

MrChromebox, 2024-06-07:
are you sure? HP's specs suggest otherwise

rosanyadav08, 2024-06-07:
I am not sure about it but i can share you its motherboards image

rosanyadav08, 2024-06-07:
Actually i new in this field , i don't have much idea

MrChromebox, 2024-06-07:
> I am not sure about it but i can share you its motherboards image

please do

rosanyadav08, 2024-06-07:
[image]

[image]

rosanyadav08, 2024-06-09:
Can you say what to do next

MrChromebox, 2024-06-09:
do you have a SuzyQ cable for debugging? If not, nothing I can do

MrChromebox, 2024-10-14:
closing as not a script issue

closed 2024-10-14 by MrChromebox