MrChromebox/scripts #409: `flashrom` fork source code

d-e-s-o, 2024-06-11:
The scripts download some `flashrom` binary blob [here](https://github.com/MrChromebox/scripts/blob/97c8cde85cb44c655a2f46984d81e75c54d0c2d7/functions.sh#L241).

Where can the source code be found? I went over your repositories but perhaps I missed it. My understanding is that it's a fork of https://flashrom.org/, which is licensed under GPL v2, so it must be out there somewhere, right? Thanks in advance for the pointer!

MrChromebox, 2024-06-11:
it's built from Google's chromium fork. I don't have a commit ID though.

d-e-s-o, 2024-06-11:
Thanks for the quick response. I think I found it:

> flashrom MrChromebox-1.4.0-devel_2024.04.18 (git:v1.2-1418-ge5ed0c6340) on Linux 5.4.80 (x86_64)

https://chromium.googlesource.com/chromiumos/third_party/flashrom

```
commit e5ed0c6340961594dcf1a22f9907d91d576b4885
Author: Maximilian Brune <maximilian.brune@9elements.com>
Date:   2024-04-11 21:05:26 +0200

    util/list_yet_unsupported_chips.h: Fix path

    Change-Id: Iecb6cf3d1f214102a243a3ffa8d0c9301263af0a
    Signed-off-by: Maximilian Brune <maximilian.brune@9elements.com>
    Reviewed-on: https://review.coreboot.org/c/flashrom/+/81855
    Tested-by: build bot (Jenkins) <no-reply@coreboot.org>
    Reviewed-by: Anastasia Klimchuk <aklm@chromium.org>
```

Could you share how you configure/built it, though? When I built it from source the binary always behaves differently, suggesting to me that I am lacking some config options being set or the like.

MrChromebox, 2024-06-11:
> When I built it from source the binary always behaves differently

I don't know what you mean by this

d-e-s-o, 2024-06-11:
> > When I built it from source the binary always behaves differently
> 
> I don't know what you mean by this

When I compile it from source I am always asked to specify a "programmer" on execution (`--programmer` or whatever the flag is called), but not so with your binary.

MrChromebox, 2024-06-12:
> When I compile it from source I am always asked to specify a "programmer" on execution (--programmer or whatever the flag is called), but not so with your binary.

there's a build option to set a default programmer if none is specified, my binary was built with it set to the internal programmer. My script still uses `-p internal` anyway though

d-e-s-o, 2024-06-12:
Awesome. Thank you!

closed 2024-06-12 by d-e-s-o