MrChromebox/scripts #410: Suspend not working on Nissa (PUJJO)

Cruzy22k, 2024-06-13:
Suspend currently doesn't work on Full-Rom/UEFI. Here is some logs, I don't really know if this enough, it shows a problem with `cros_ec` process...

`cruzy-Pujjo ModemManager[1089]: <msg> [sleep-monitor-systemd] system is about to suspend
2024-06-02T05:47:58.591257+10:00 cruzy-Pujjo NetworkManager[935]: <info>  [1717271278.5889] manager: sleep: sleep requested (sleeping: no  enabled: yes)
2024-06-02T05:47:58.591334+10:00 cruzy-Pujjo NetworkManager[935]: <info>  [1717271278.5890] device (p2p-dev-wlp0s20f3): state change: disconnected -> unmanaged (reason 'sleeping', sys-iface>
2024-06-02T05:47:58.591381+10:00 cruzy-Pujjo NetworkManager[935]: <info>  [1717271278.5893] manager: NetworkManager state is now ASLEEP
2024-06-02T05:47:58.591432+10:00 cruzy-Pujjo NetworkManager[935]: <info>  [1717271278.5893] device (wlp0s20f3): state change: activated -> deactivating (reason 'sleeping', sys-iface-state: >
2024-06-02T05:47:58.593321+10:00 cruzy-Pujjo google-chrome.desktop[2326]: [2319:2387:0602/054758.592826:ERROR:connection_factory_impl.cc(483)] ConnectionHandler failed with net error: -2
2024-06-02T05:47:58.597753+10:00 cruzy-Pujjo dbus-daemon[813]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service>`

[1]: Reached target sleep.target - Sleep.
2024-06-02T06:23:31.432996+10:00 cruzy-Pujjo systemd[1]: Starting systemd-suspend.service - System Suspend...
2024-06-02T06:23:31.469565+10:00 cruzy-Pujjo kernel: PM: suspend entry (s2idle)
2024-06-02T06:23:31.469742+10:00 cruzy-Pujjo systemd-sleep[3584]: Performing sleep operation 'suspend'...
2024-06-02T06:23:31.486955+10:00 cruzy-Pujjo kernel: Filesystems sync: 0.015 seconds
2024-06-02T06:23:31.496378+10:00 cruzy-Pujjo kernel: Freezing user space processes
2024-06-02T06:23:47.154966+10:00 cruzy-Pujjo kernel: Freezing user space processes completed (elapsed 0.003 seconds)
2024-06-02T06:23:47.155139+10:00 cruzy-Pujjo kernel: OOM killer disabled.
2024-06-02T06:23:47.155141+10:00 cruzy-Pujjo kernel: Freezing remaining freezable tasks
2024-06-02T06:23:47.155143+10:00 cruzy-Pujjo kernel: Freezing remaining freezable tasks completed (elapsed 0.000 seconds)
2024-06-02T06:23:47.155144+10:00 cruzy-Pujjo kernel: printk: Suspending console(s) (use no_console_suspend to debug)
2024-06-02T06:23:47.155144+10:00 cruzy-Pujjo kernel: ACPI: EC: interrupt blocked
2024-06-02T06:23:47.155192+10:00 cruzy-Pujjo kernel: ACPI: EC: interrupt unblocked
2024-06-02T06:23:47.155193+10:00 cruzy-Pujjo kernel: i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.20.0
2024-06-02T06:23:47.155193+10:00 cruzy-Pujjo kernel: i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
2024-06-02T06:23:47.155194+10:00 cruzy-Pujjo kernel: i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads
2024-06-02T06:23:47.155194+10:00 cruzy-Pujjo kernel: i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
2024-06-02T06:23:47.155195+10:00 cruzy-Pujjo kernel: i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
2024-06-02T06:23:47.155195+10:00 cruzy-Pujjo kernel: i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
2024-06-02T06:23:47.155196+10:00 cruzy-Pujjo kernel: ------------[ cut here ]------------
2024-06-02T06:23:47.155196+10:00 cruzy-Pujjo kernel: EC detected sleep transition timeout. Total sleep transitions: 0
2024-06-02T06:23:47.155197+10:00 cruzy-Pujjo kernel: WARNING: CPU: 3 PID: 3584 at drivers/platform/chrome/cros_ec.c:143 cros_ec_sleep_event+0x12f/0x150 [cros_ec]
2024-06-02T06:23:47.155198+10:00 cruzy-Pujjo kernel: Modules linked in: ccm rfcomm snd_seq_dummy snd_hrtimer qrtr snd_soc_sof_rt5682 snd_soc_intel_sof_board_helpers snd_soc_intel_hda_dsp_co>
2024-06-02T06:23:47.155201+10:00 cruzy-Pujjo kernel:  cros_ec_sensorhub cros_ec_debugfs soundwire_bus cros_usbpd_logger cros_usbpd_charger cros_ec_chardev iwlmvm snd_soc_rt5682 snd_soc_rl62>
2024-06-02T06:23:47.155257+10:00 cruzy-Pujjo kernel:  snd_seq_device i2c_algo_bit snd_timer processor_thermal_mbox igen6_edac snd soundcore cros_usbpd_notify cros_kbd_led_backlight chromeos>
2024-06-02T06:23:47.155259+10:00 cruzy-Pujjo kernel: CPU: 3 PID: 3584 Comm: systemd-sleep Not tainted 6.8.0-31-generic #31-Ubuntu
2024-06-02T06:23:47.155260+10:00 cruzy-Pujjo kernel: Hardware name: Google Pujjo/Pujjo, BIOS MrChromebox-4.22.4 04/16/2024
2024-06-02T06:23:47.155260+10:00 cruzy-Pujjo kernel: RIP: 0010:cros_ec_sleep_event+0x12f/0x150 [cros_ec]
2024-06-02T06:23:47.155274+10:00 cruzy-Pujjo kernel: Code: 0f 87 e6 08 00 00 83 e3 01 75 a0 8b 75 dc 48 c7 c7 00 c2 5f c0 89 45 c4 c6 05 74 1e 00 00 01 81 e6 ff ff ff 7f e8 91 7a 70 e7 <0f>>
2024-06-02T06:23:47.155276+10:00 cruzy-Pujjo kernel: RSP: 0018:ffffa75245e63be0 EFLAGS: 00010246
2024-06-02T06:23:47.155276+10:00 cruzy-Pujjo kernel: RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
2024-06-02T06:23:47.155276+10:00 cruzy-Pujjo kernel: RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
2024-06-02T06:23:47.155277+10:00 cruzy-Pujjo kernel: RBP: ffffa75245e63c20 R08: 0000000000000000 R09: 0000000000000000
2024-06-02T06:23:47.155278+10:00 cruzy-Pujjo kernel: R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000000
2024-06-02T06:23:47.155278+10:00 cruzy-Pujjo kernel: R13: 0000000000000001 R14: ffffffffa85e9d40 R15: ffff8c36816d2090
2024-06-02T06:23:47.155279+10:00 cruzy-Pujjo kernel: FS:  00007f7a33b99440(0000) GS:ffff8c37f7b80000(0000) knlGS:0000000000000000
2024-06-02T06:23:47.155279+10:00 cruzy-Pujjo kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
2024-06-02T06:23:47.155280+10:00 cruzy-Pujjo kernel: CR2: 00005fc2eeb2bc86 CR3: 00000001cdc90000 CR4: 0000000000f50ef0
2024-06-02T06:23:47.155280+10:00 cruzy-Pujjo kernel: PKRU: 55555554
2024-06-02T06:23:47.155280+10:00 cruzy-Pujjo kernel: Call Trace:
2024-06-02T06:23:47.155281+10:00 cruzy-Pujjo kernel:  <TASK>
2024-06-02T06:23:47.155281+10:00 cruzy-Pujjo kernel:  ? show_regs+0x6d/0x80
2024-06-02T06:23:47.155282+10:00 cruzy-Pujjo kernel:  ? __warn+0x89/0x160
2024-06-02T06:23:47.155282+10:00 cruzy-Pujjo kernel:  ? cros_ec_sleep_event+0x12f/0x150 [cros_ec]
2024-06-02T06:23:47.155282+10:00 cruzy-Pujjo kernel:  ? report_bug+0x17e/0x1b0

DMESG 

[   92.104901] CPU: 2 PID: 2365 Comm: systemd-sleep Not tainted 6.8.0-31-generic #31-Ubuntu
[   92.104903] Hardware name: Google Pujjo/Pujjo, BIOS MrChromebox-4.22.4 04/16/2024
[   92.104905] RIP: 0010:cros_ec_sleep_event+0x12f/0x150 [cros_ec]
[   92.104909] Code: 0f 87 e6 08 00 00 83 e3 01 75 a0 8b 75 dc 48 c7 c7 00 42 4c c0 89 45 c4 c6 05 74 de 00 00 01 81 e6 ff ff ff 7f e8 91 ba 84 c8 <0f> 0b 8b 45 c4 e9 77 ff ff ff c7 45 d4 04 00 00 00 e9 35 ff ff ff
[   92.104911] RSP: 0018:ffffc2c7c2993b28 EFLAGS: 00010246
[   92.104913] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
[   92.104918] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[   92.104919] RBP: ffffc2c7c2993b68 R08: 0000000000000000 R09: 0000000000000000
[   92.104920] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000000
[   92.104920] R13: 0000000000000001 R14: ffffffff895e9d40 R15: ffffa0e8c16d7090
[   92.104922] FS:  0000753688745440(0000) GS:ffffa0ea37b00000(0000) knlGS:0000000000000000
[   92.104923] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[... 46 lines cut ...]
[   92.105073] R10: 0000000000000001 R11: 0000000000000202 R12: 0000000000000004
[   92.105074] R13: 00005800aa06a460 R14: 00005800aa0662d0 R15: 0000753688a01ee0
[   92.105076]  </TASK>
[   92.105076] ---[ end trace 0000000000000000 ]---
[   92.105494] OOM killer enabled.
[   92.105495] Restarting tasks ... 
[   92.105539] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[   92.106934] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[   92.111733] done.
[   92.111746] random: crng reseeded on system resumption
[   92.112483] thermal thermal_zone6: failed to read out thermal zone (-61)
[   92.112719] PM: suspend exit
[   92.270577] iwlwifi 0000:00:14.3: WRT: Invalid buffer destination
[   92.424314] iwlwifi 0000:00:14.3: WFPM_UMAC_PD_NOTIFICATION: 0x20
[   92.424373] iwlwifi 0000:00:14.3: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[   92.424435] iwlwifi 0000:00:14.3: WFPM_AUTH_KEY_0: 0x90
[   92.424497] iwlwifi 0000:00:14.3: CNVI_SCU_SEQ_DATA_DW9: 0x0
[   92.425857] iwlwifi 0000:00:14.3: RFIm is deactivated, reason = 4
[   96.313913] wlp0s20f3: authenticate with 64:fd:96:b7:ef:cf (local address=a0:02:a5:aa:c4:ef)
[   96.315021] wlp0s20f3: send auth to 64:fd:96:b7:ef:cf (try 1/3)
[   96.348581] wlp0s20f3: authenticated
[   96.349593] wlp0s20f3: associate with 64:fd:96:b7:ef:cf (try 1/3)
[   96.351143] wlp0s20f3: RX AssocResp from 64:fd:96:b7:ef:cf (capab=0x1011 status=0 aid=2)

Yes I have tried other operating systems like feodora and linux mint

Cruzy22k, 2024-06-13:
If anyone has any ideas as to what is the cause, or if I posted this in the wrong place, please help

MrChromebox, 2024-06-13:
sounds like it's a Linux kernel/driver issue not a firmware one. Pujjo/Pujjoflex have no issues with suspend/resume under Windows. I'll try and replicate the issue here when I have some time. please let me know the distro/kernel and provide a dmesg log after a fresh boot

Cruzy22k, 2024-06-13:
It was Ubuntu 24.04 and i also tested it on Fedora. I am afraid that I re-installed chromeOS just then...I can re-install Linux if you want, tbh I didn't expect such a quick response.

Cruzy22k, 2024-06-13:
Yes windows resume/suspend worked perfectly...

care95, 2024-06-22:
what the zest

Cruzy22k, 2024-06-23:
Bruh