MrChromebox/scripts #412: `flashrom` and `cbfstool` provided by shell scripts aren't compatible with Musl.

GamePlayer-8, 2024-06-16:
Good afternoon,

There's an issue with latest scripts, leading to error:
```shell
MrChromebox Firmware Utility Script starting up

Downloading supporting files...

Downloading required tools...

Getting device/system info...

Flashrom is unable to read current firmware; cannot continue:

./functions.sh: line 456: /tmp/flashrom: cannot execute: required file not found

You may need to add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu 23.04+).

See https://www.flashrom.org/FAQ for more info.
```

The same error occurs with `cbfstool`, but `gbb_utility` runs fine.
The system is Alpine Linux v3.20.

MrChromebox, 2024-06-16:
not sure there's much I can do there, other than recommend you live boot another distro to update

GamePlayer-8, 2024-06-16:
I may suggest using `staticx` for turning the dynamic binary into a static one without the need of recompiling it. Not sure if that would work in that case, but the command execution needs to be done on the LibC under what the dynamic binary was compiled.

larshint, 2024-07-18:
flashrom and cbfstool compile fine on musl systems, so it is possible to just modify the scripts to not use distributed binaries.

Anyway, I just tested [staticx](https://github.com/JonathonReinhart/staticx) (interesting project, I wasn't aware of it). It works.
Compilation took place in Linux Mint (glibc-based), tested in Gentoo (musl-based).

Reproducible steps (Linux Mint):

Install staticx:
```console
# apt-get install pip patchelf
$ pip install staticx
```
Compile flashrom\cbfstool as described [here](https://wiki.gentoo.org/wiki/MrChromebox%27s_coreboot#VPD_injection).

Bundle all dependencies into one executable file:
```console
$ ~/.local/bin/staticx <glibc-linked file> <static file name>
```
While this works, bundling can cause license issues.

Dependencies of flashrom:
```console
$ lddtree flashrom
glibcflashrom => ./glibcflashrom (interpreter => /lib64/ld-linux-x86-64.so.2)
    libpci.so.3 => /lib/x86_64-linux-gnu/libpci.so.3
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1
        libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1
            ld-linux-x86-64.so.2 => /lib64/ld-linux-x86-64.so.2
    libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6
```

Dependencies of cbfstool:
```console
$ lddtree glibccbfstool
glibccbfstool => ./glibccbfstool (interpreter => /lib64/ld-linux-x86-64.so.2)
    libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6
        ld-linux-x86-64.so.2 => /lib64/ld-linux-x86-64.so.2
```
Each license of each dependency must be analyzed and included with all requirements. It is also worth noting that the licenses may not be compatible. Since this project does not control these dependencies, they may change over time. And each time it needs to be analyzed. Plus the libraries may have vulnerabilities, if they are statically linked, there will be no updates. In my opinion, the risk is high.

MrChromebox, 2024-11-16:
musl support has been added

closed 2024-11-16 by MrChromebox