MrChromebox/scripts #413: minor script issue

nathblade16, 2024-06-18:
I own an asus c213sa on ueifi bios, I updated its uefi bios to the latest version and it downloaded the firmware for Acer Chromebook Spin 11 (R751T) (ELECTRO) even though it should download it for the asus c213sa (REEF), it isn't causing any issues for me other than it reporting the wrong model, can you fix this?

nathblade16, 2024-06-18:
> I've updated the HWID handling for reef/electro to try and better handle distinction between the two models, but it is purely a cosmetic issue - the same firmware image is used for both devices.
> 
> also, since your device is now flagged as an electro, not sure how easily we can switch it over automatically unless I add a prompt to the script to select between the two

yeah, don't really care... I would prefer it to show my correct model but its honestly kind of funny seeing the chromebook think its now an electro/acer spin 11

MrChromebox, 2024-06-19:
> yeah, don't really care... I would prefer it to show my correct model but its honestly kind of funny seeing the chromebook think its now an electro/acer spin 11

if you want to change, I can provide instructions to flash it manually

nathblade16, 2024-06-19:
> > yeah, don't really care... I would prefer it to show my correct model but its honestly kind of funny seeing the chromebook think its now an electro/acer spin 11
> 
> if you want to change, I can provide instructions to flash it manually

I would like instructions but I might decide not to if its too hard

MrChromebox, 2024-06-19:
run the script from Linux to the menu (so it downloads all the tools needed), then exit. Then run the following commands in a terminal:

`cd /tmp`
`curl -LO https://mrchromebox.tech/files/firmware/full_rom/coreboot_edk2-reef-mrchromebox_20240615.rom`
`sudo ./flashrom -p internal -w coreboot_edk2-reef-mrchromebox_20240615.rom --fmap -i COREBOOT --noverify-all`
`sudo reboot`

nathblade16, 2024-06-19:
> run the script from Linux to the menu (so it downloads all the tools needed), then exit. Then run the following commands in a terminal:
>
> `cd /tmp` `curl -LO [...]

thank you, it worked

closed 2024-06-19 by nathblade16