MrChromebox/scripts #415: Image size doesn't match the expected size

Shinji321-123, 2024-06-20:
Hello, I have a problem when trying to Install Full ROM Firmware on my Chromebook, it puts me an error saying :

Error : Image size (8388608 B) doesn't match the expected size (16777216 B) !
FAILED

MrChromebox, 2024-06-20:
I'm going to need to know what device you're using, and what file name the script is downloading

Shinji321-123, 2024-06-20:
Thanks for responding, it's an Asus Chromebook CX1400CNA, for the file name where can i see it ?

MrChromebox, 2024-06-20:
before you press a key to see the flashrom error, the script shows the name of the file it's downloading as part of the flashing process. Also show me the script main menu screen please

Shinji321-123, 2024-06-20:
Here they are
[image]
[image]

MrChromebox, 2024-06-20:
so here is the problem -- you or someone else changed the HWID on the device, and now the script thinks you have a different device than you actually have. Why was that done? SMH. Good way to potentially brick your device.

So, let's do this:
- select option 4 - set HWID
- set the HWID (exactly) to: RABBID EFA-123
- reboot
- re-run the script
- verify your device is identified correctly by the script in the menu.

I also don't understand why you'd run the script, see that it misidentifies your device as a Toshiba Chromebook 2, and proceed anyway. Sigh.

Shinji321-123, 2024-06-20:
Maybe I made an error while reinstalling ChromeOS on my computer, but now it works, thank you so much !

closed 2024-06-20 by MrChromebox