MrChromebox/scripts #416: pixelbook go restore

flymzero, 2024-06-23:
i want restore but get problem

[image]

nathblade16, 2024-06-24:
DO NOT REBOOT YOUR CHROMEBOOK, you should probably put uefi rom back on there so you can restart and have it actually turn on and after a bit mrchromebox might respond, this is just advice from someone who knows a bit about these scripts and stuff and can't fully resolve this issue

MrChromebox, 2024-06-24:
agreed, I would reflash the UEFI firmware here before rebooting

zzup110, 2024-07-30:
I also encountered this problem on Pixelbook go(2019) ATLAS

flymzero, 2024-07-31:
> I also encountered this problem on Pixelbook go(2019) ATLAS

I use another computer to compulsively brush bios

closed 2024-07-31 by flymzero

clearc352, 2025-07-08:
> > I also encountered this problem on Pixelbook go(2019) ATLAS
> 
> I use another computer to compulsively brush bios
can you provide some details? I have the same problem.