MrChromebox/scripts #417: acer chromebox cxi3 freezes when plugging in USB-C device

salamanders, 2024-06-24:
(Unsure if this would be a firmware issue or not!  Might just be damaged hardware)

1. Great instructions on the website, thank you!
2. I was able to convert an acer chromebox cxi3 to 100% linux bootup
3. It seems ok, *except* when I plug in a USB-C device (this happens to be a webcam).  When I plug it in, the device freezes.

@chromebox:~$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0a2a Intel Corp. Bluetooth wireless interface
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

Fully up to date with latest 2024-05 firmware script.

Any chance it sees a new thing being plugged in, and says "oh, cool, now I'm getting power from USB-C instead of from the dedicated power barrel plug, I'll switch over" - and then it keels over?  It sounds like the fan turns off when I plug in the USB-C cam.

MrChromebox, 2024-06-24:
if you have a SuzyQ cable/adapter, can connect to the EC console and see what's happening there when the webcam gets plugged in. Or try a USB-C to A adapter and see if it works on a regular USB3 port?

salamanders, 2024-06-25:
> Or try a USB-C to A adapter and see if it works on a regular USB3 port?

It does!  I found a USB-C to A cable (blue, so I'm hoping it is USB 3.0), and it worked fine.  So it isn't like a short in the camera is suddenly grabbing all the power.  TBD if the speed is enough, but the camera runs fine, so the usb A power is sufficient to run the camera.  Yay!

If there isn't a standard log file I can upload, I'm guessing this is too rare a condition to be worth chasing down - given that it crashes as soon as I plug in the C-to-C cable, it makes it very hard to diagnose.

closed 2024-06-25 by salamanders

MrChromebox, 2024-06-25:
given that the only real way to debug it would be with a SuzyQ cable, which also occupies the USB-C port, this would be a difficult issue to debug without a servo debug interface