MrChromebox/scripts #420: SCRIPT NOT WORKING

Ismo404, 2024-07-25:
I was trying to restore my chromebook to its original state, i got an Asus c423na-eb0287.
The problem comes when i boot my pc into the live usb and run `cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh` it just give me an error saying that it couldn’t connect to chromebox.tech server even though i was connected to my wifi.
does anyone know how to resolve that?

screenshot: 
[image]

clm1919, 2024-07-25:
does your live USB have a firewall on by default?  it might be blocking port 80

what happens when you try  "ping mrchromebox.tech"

MrChromebox, 2024-07-25:
> does anyone know how to resolve that?

typos are a bitch

closed 2024-07-25 by MrChromebox