MrChromebox/scripts #421: Can't Run Script on Arch Linux

joseyskindaweird, 2024-07-25:
MrChromebox Firmware Utility Script starting up
firmware-util.sh: line 63: /home/strange/sources.sh: No such file or directory
firmware-util.sh: line 64: /home/strange/firmware.sh: No such file or directory
firmware-util.sh: line 65: /home/strange/functions.sh: No such file or directory
firmware-util.sh: line 71: prelim_setup: command not found
firmware-util.sh: line 75: diagnostic_report_save: command not found
MrChromebox Firmware Utility setup was unsuccessful
 * diagnosics report has been saved to /tmp/mrchromebox_diag.txt
 * go to https://forum.chrultrabook.com/ for help

this the output, any fixes/help?

MrChromebox, 2024-07-25:
don't use snap-based curl? That's the only thing I can think of that would cause it to try and download the files to your home dir instead of /tmp

joseyskindaweird, 2024-07-25:
> don't use snap-based curl? That's the only thing I can think of that would cause it to try and download the files to your home dir instead of /tmp

i'm not using the snap version, i'm using the one from arch's repo.

MrChromebox, 2024-07-26:
> i'm not using the snap version, i'm using the one from arch's repo.

well I don't use Arch and haven't heard of anyone else having this issue.

How exactly are you running the script? following directions on my site?

joseyskindaweird, 2024-07-26:
> > i'm not using the snap version, i'm using the one from arch's repo.
> 
> 
> 
> well I don't use Arch and haven't heard of anyone else having this issue.
> 
> 
> 
> How exactly are you running the script? following directions on my site?

mhm, i'm using the 
'cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh' 
command

MrChromebox, 2024-07-26:
it's something with curl as installed by Arch then? IDK. I'd say boot another Live USB that's know to work (ie, Ubuntu 2404) or try getting help on the Arch forums.

joseyskindaweird, 2024-07-26:
> it's something with curl as installed by Arch then? IDK. I'd say boot another Live USB that's know to work (ie, Ubuntu 2404) or try getting help on the Arch forums.

ill boot a live ubuntu install once my main rig comes back.

cmdada, 2024-08-29:
I haven't had this problem, and on a fully updated arch system the utility script runs fine

LZeugirdor, 2024-09-20:
Can confirm I am also having this issue on Linux Mint 22

`MrChromebox Firmware Utility Script starting up
firmware-util.sh: line 63: /home/lzeugirdor/sources.sh: No such file or directory
firmware-util.sh: line 64: /home/lzeugirdor/firmware.sh: No such file or directory
firmware-util.sh: line 65: /home/lzeugirdor/functions.sh: No such file or directory
firmware-util.sh: line 71: prelim_setup: command not found
firmware-util.sh: line 75: diagnostic_report_save: command not found
MrChromebox Firmware Utility setup was unsuccessful
 * diagnosics report has been saved to /tmp/mrchromebox_diag.txt
 * go to https://forum.chrultrabook.com/ for help`

LZeugirdor, 2024-09-20:
Also there is no log under /tmp/mrchromebox_diag.txt
I was monitoring the directory and it seems like nothing is being written to /tmp,

LZeugirdor, 2024-09-21:
One last thing, I know this is taboo but entering root via sudo su allows the script to run normally, not sure what else to do otherwise.

MrChromebox, 2024-09-21:
> Can confirm I am also having this issue on Linux Mint 22
>
> ```
> MrChromebox Firmware Utility Script starting up
> firmware-util.sh: line 63: /home/lzeugirdor/sources.sh: No such file or directory [...]

that's because your distro's curl app is sandboxed and is not downloading the files where the script is telling it to

LZeugirdor, 2024-09-22:
> that's because your distro's curl app is sandboxed and is not downloading the files where the script is telling it to

Interesting. Is this an apparmor issue?

MrChromebox, 2024-09-22:
> > that's because your distro's curl app is sandboxed and is not downloading the files where the script is telling it to
> 
> Interesting. Is this an apparmor issue?

no idea, don't use AA and don't use Arch. I've only seen this behavior before on Ubuntu-based distros which use snap-installed curl (vs apt-installed) since Snap sandboxes things

LZeugirdor, 2024-09-29:
I promise I don't use snap installed stuff, Mint disables snap explicitly unless they've changed something recently. The only thing I can think of is apparmor which I think most mainline distros install.

cameronj86, 2024-10-29:
Experienced the same issue on Debian 12 using curl from the Debian repo. 

I ran the `firmware-util.sh` and the error through Claude, and the issue appears to be related to whether the user has a .git file or not. I just wiped my chrultrabook and so git isn't populated yet, and I believe this line does it:
`if [ ! -d "$script_dir/.git" ]; then
    script_dir="."` 

This temporary workaround did the trick:
```
cd /tmp && curl -LO https://mrchromebox.tech/firmware-util.sh && \
curl -LO https://raw.githubusercontent.com/MrChromebox/scripts/main/sources.sh && \
curl -LO https://raw.githubusercontent.com/MrChromebox/scripts/main/functions.sh && \
curl -LO https://raw.githubusercontent.com/MrChromebox/scripts/main/firmware.sh && \
sudo bash firmware-util.sh
```

cameronj86, 2024-10-29:
> @cameronj86 sounds like curl is broken then, maybe I should use wget instead

Whoops, I think I worded things poorly or I don't fully comprehend. 

Looking at the bash, my understanding is that the script looks for a .git file in $script_dir, and failing that, tries to write locally. Users that don't have a .git in $script_dir (/usr/bin/ for me when I step through the code) will get the error code. AFAICT, it seems to be outside of curl, but I might be misreading things.

MrChromebox, 2024-10-29:
> Looking at the bash, my understanding is that the script looks for a .git file in $script_dir, and failing that, tries to write locally.

that's not correct. The script looks for a .git **directory** in the directory that firmware-util.sh was run from, to determine if it's being run from a cloned repo. If so, it doesn't download the additional .sh files, but uses the ones in $script_dir. If it's not being run from a repo, then it downloads those files to /tmp.

If curl is doing naughty things and sandboxing downloads, then the files don't actually end up in /tmp/, and then firmware-util.sh is unable to source them

cameronj86, 2024-10-29:
Appreciate the additional context, thanks for explaining and good documentation in general

LZeugirdor, 2025-12-10:
Renamed .git to git, script functions normally. Can this be included as a warning in the script before attempting to launch so that confusion may be prevented?