MrChromebox/scripts #425: After enabling legacy boot, ctrl+L shows two non-functional options.

ddowen, 2024-08-16:
Motherboard: HP BARLA

HWID: BARLA C4B-A4L-F3V-A8D-52Q-C2E-A9T

problem: when I press ctrl+l it shows 2 options, “U-boot” and “TianoCore”. 

Choosing U-boot with Debian 12 live boot flashed on the microSD, yields a terminal style interface that fails to finish with the error “not Z-image”

Choosing TianoCore freezes imminently and fails to unfreeze.

I did check the FAQ and found no fix.

does anyone have any way to fix this?

MrChromebox, 2024-08-17:
there is no fix, that why it's noted on the supported devices page. Nothing I can do about it unfortunately

closed 2024-08-17 by MrChromebox

ddowen, 2024-08-17:
> there is no fix, that why it's noted on the supported devices page. Nothing I can do about it unfortunately

It is listed under supported devices for both full ROM and legacy, or is that outdated? I checked earlier today

MrChromebox, 2024-08-17:
https://docs.mrchromebox.tech/docs/known-issues.html