MrChromebox/scripts #426: error in download & backup

mehdiyp, 2024-08-18:
hi 
This message is displayed when it reaches the stage of downloading the firmware and installing it on the USB:
./firmware.sh: line 781: [[:  1): syntex error in expression (error token is ")")
./firmware.sh: line 779: [[:  1): syntex error in expression (error token is ")")
./firmware.sh: line 786: [[:  1): syntex error in expression (error token is ")")

and exits the script.

What should I do?

MrChromebox, 2024-08-19:
provide more detail? Perhaps something like "I'm running <distro name and version> from <internal storage or usb> and am attempting to run the Firmware Utility Script as such: <command line used>"

Also, the script is not installed anywhere. it's downloaded and run from local storage, and anything it downloads subsequently is put in /tmp and so disappears upon reboot.

mehdiyp, 2024-08-19:
As you can see in the attached photo;I enter the VT2 terminal and the command cd; I run curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh, the script is executed, I choose number 2 and answer the questions.
then I select usb device and get the error message I wrote above.
Chromebook lenovo 300e
[image]

MrChromebox, 2024-08-19:
SMH. The script says to enter a number. `1` is a number, `1)` is not.

closed 2024-08-19 by MrChromebox

mehdiyp, 2024-08-19:
Yes, the problem is solved, thank you!