MrChromebox/scripts #429: Pixel Slate Restore Stock Firmware issue: "Error: Image size (267 B) doesn't match the expected size (16777216 B)! Restoring PCI config space for 00:1f:5 reg 0xdc

HAOJINGHAO, 2024-10-16:
Hi MrChromebox, thanks for the amazing work. I recently tried restoring the stock firmware for Google Pixel Slate and ran into an issue: "Error: Image size (267 B) doesn't match the expected size (16777216 B)! Restoring PCI config space for 00:1f:5 reg 0xdc

Steps to reproduce: run script -> 2) Restore Stock Firmware -> Continue=Y -> firmware backup=N -> Confirm model=Y

Here is the full output from the script, thanks again for the help:

_Do you wish to continue? [y/N] y

Do you have a firmware backup file on USB? [y/N] n

That's ok, I'll download a shellball firmware for you.

Confirm system details:

Device: Google Pixel Slate
Board Name: NOCTURNE

? [y/N] y

Downloading shellball.nocturne.bin

VPD extracted from current firmware

Merging VPD into shellball/recovery image firmware

Injecting HWID into shellball/recovery image firmware

Restoring stock firmware

flashrom MrChromebox-1.4.0-devel_2024.04.18 (git:v1.2-1418-ge5ed0c6340) on Linux 6.8.0-45-generic (x86_64)
flashrom was built with GCC 11.4.0, little endian
Command line (10 args): /tmp/flashrom -p internal:boardmismatch=force --ifd -i bios -N -w /tmp/stock-firmware.rom -o /tmp/flashrom.log
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Initializing internal programmer
Found candidate at: 00000500-00000528
Found coreboot table at 0x00000500.
Found candidate at: 00000000-000004e4
Found coreboot table at 0x00000000.
coreboot table found at 0x7aa21000.
coreboot header(24) checksum: b186 table(1228) checksum: b7d6 entries: 41
Vendor ID: Google, part ID: Nocturne
Using Internal DMI decoder.
DMI chassis-type is not specific enough.
DMI string system-manufacturer: "Google"
DMI string system-product-name: "Nocturne"
DMI string system-version: "1.0"
DMI string baseboard-manufacturer: "Google"
DMI string baseboard-product-name: "Nocturne"
DMI string baseboard-version: "1.0"
Found chipset "Intel Kaby Lake Y w/ iHDCP2.2 Prem." with PCI ID 8086:9d4b.
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... BIOS_SPI_BC = 0x8b: BIOS Interface Lock-Down: enabled, Boot BIOS Straps: 0x0 (SPI)
Top Swap: not enabled
SPI Read Configuration: prefetching enabled, caching enabled, 
BIOS_CNTL = 0x8b: BIOS Lock Enable: enabled, BIOS Write Enable: enabled
Warning: Setting BIOS Control at 0xdc from 0x8b to 0x89 failed.
New value is 0x8b.
SPIBAR = 0x00007a4075e1a000 (phys = 0xfe010000)
0x04: 0xf800 (HSFS)
HSFS: FDONE=0, FCERR=0, AEL=0, SCIP=0, PRR34_LOCKDN=1, WRSDIS=1, FDOPSS=1, FDV=1, FLOCKDN=1
SPI Configuration is locked down.
Reading OPCODES... done
        OP        Type      Pre-OP
op[0]: 0x01, write w/o addr, none
op[1]: 0x02, write w/  addr, none
op[2]: 0x03, read  w/  addr, none
op[3]: 0x05, read  w/o addr, none
[... 202 lines cut ...]
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
HSFC: FGO=1, FCYCLE=0, WET=0, FDBC=63, SME=0
done.
Assuming chipset '100 series Sunrise Point'.
Added layout entry 00000000 - 00000fff named fd
Added layout entry 00200000 - 00ffffff named bios
Added layout entry 00001000 - 001fffff named me
Using region: "bios".
Error: Image size (267 B) doesn't match the expected size (16777216 B)!
Restoring PCI config space for 00:1f:5 reg 0xdc
An error occurred restoring the stock firmware. DO NOT REBOOT!

Press [Enter] to return to the main menu.

MrChromebox, 2024-10-16:
this isn't a flashing error, I don't offer shellballs anymore -- I need to remove the script option

MrChromebox, 2024-10-16:
script updated, shellball option removed. You can restore the stock firmware from a backup or from a recovery image

closed 2024-10-16 by MrChromebox