MrChromebox/scripts #430: It is impossible to return to the standard firmware

Vany913, 2024-10-27:
I want to go back to stock firmware, I lost my old firmware. Here are my specs.

[image]

I used the script in the sequence:
 Restore Stock Firmware>yes>yes>use recovery usb>my flash

result in the photo :(

[image]

MrChromebox, 2024-10-27:
Going back to stock firmware on devices which are past EOL is not supported by the script and not recommended at all.

What are you trying to accomplish exactly?

Vany913, 2024-10-27:
I understand that my device is not supported. I want to go back to factory firmware to install rw_legacy. But I can't because I didn't understand how important it was to save the old firmware file😓

Vany913, 2024-10-27:
I understand that my device is not supported. I want to go back to factory firmware to install rw_legacy. But I can't, because I didn't understand how important it was to save the old firmware file😓

Vany913, 2024-10-27:
What are my options?

Vany913, 2024-10-27:
What are my options for this?

MrChromebox, 2024-10-27:
download this, extract zip, copy to USB, use script to t restore. This is the stock firmware with RW_LEGACY firmware pre-packed, and the flags set to boot to RW_LEGACY automatically.

[coreboot-Google_Ultima.7287.131.63_with_rwl.zip](https://github.com/user-attachments/files/17535218/coreboot-Google_Ultima.7287.131.63_with_rwl.zip)

Vany913, 2024-10-27:
Yes, it worked☺️ Thank you! P.S. maybe it will be useful to someone else?

Vany913, 2024-10-27:
Sorry, I was able to set rw_legacy but if I press ctrl+l nothing happens

MrChromebox, 2024-10-27:
did you perform a ChromeOS USB recovery? If so, you erased the RW_LEGACY firmware which I preloaded for you

Vany913, 2024-10-28:
I must have done everything wrong, what sequence of actions should I have done?

Vany913, 2024-10-28:
I selected the item firmware from a flash drive

MrChromebox, 2024-10-28:
> I must have done everything wrong, what sequence of actions should I have done?

restore firmware using the file I provided

reboot

that's it. the firmware is set to boot from RW_LEGACY by default

Vany913, 2024-10-28:
Can I update the file you provided according to the guide? https://docs.chrultrabook.com/docs/firmware/manually-flashing.html

MrChromebox, 2024-10-29:
if you are now able to boot ChromeOS, you can manually flash just the RW_LEGACY part of that file:

`sudo flashrom -p internal -w coreboot-Google_Ultima.7287.131.63_with_rwl.bin -i RW_LEGACY`

just make sure to use the full path to the file or copy it to the directory you run the command from

Pajamaland, 2024-11-06:
> Going back to stock firmware on devices which are past EOL is not supported by the script and not recommended at all.
> 
> What are you trying to accomplish exactly?

Hey sorry to necro this but you wouldn't have a copy of the stock firmware for the Acer Chromebook 314 (DROID) I could use to get back to stock?

MrChromebox, 2024-11-07:
> Hey sorry to necro this but you wouldn't have a copy of the stock firmware for the Acer Chromebook 314 (DROID) I could use to get back to stock?

I don't have stock firmware for any devices. The script provides an option to extract the generic stock firmware for a given board from the recovery USB. Alternatively, you can use coreboot's `crosfirmware.sh` script in `util/chromeos/` to accomplish the same thing if you need an image to flash externally.

Vany913, 2024-11-07:
Sorry for the absence, I think I downloaded the file and then rebooted. When I got a white screen that said "ChromeOS verification is disabled" I pressed Ctrl+L and it sounded 2 beeps, what does that mean?

MrChromebox, 2024-11-07:
> I pressed Ctrl+L and it sounded 2 beeps, what does that mean?

it means that dev_boot_legacy is not enabled, or that the RW_LEGACY payload is not bootable

Vany913, 2024-11-07:
what to do to fix it?

Vany913, 2024-11-07:
[image]
here is the log while flashing

MrChromebox, 2024-11-07:
looks fine. now run `sudo crossystem dev_boot_legacy=1`

if still beeps, then need to flash RW_LEGACY differently I guess

Vany913, 2024-11-07:
I was offered: PLEASE USE 'dev_boot_altfw' INSTEAD OF 'dev_boot_legacy' I did and it seems to have worked, ctrl+l now brings up a window I didn't see before and if you don't do anything it hangs with a black screen. And if you try again and then when you press esc you will be able to select the boot device. Should it be so?

MrChromebox, 2024-11-07:
yes?

MrChromebox, 2024-11-07:
as long as SeaBIOS is loading via RW_LEGACY, then I would use the firmware utility script under ChromeOS to set the GBB flags to legacy boot + 1s default, then boot and install your OS

MrChromebox, 2024-11-07:
> I installed Ubuntu 24.04 to test and everything seems to work :). Can you write point by point what to do to make my OS default. P.S. on full firmware (uefi) i couldn't install linux if the kernel is newer than 4.15 and after a while the system froze and wouldn't boot anymore.

run my script from linux, and select the option to set the boot flags. set to legacy + 1s default.

Vany913, 2024-11-07:
I have no way to set download marks.
[image]

MrChromebox, 2024-11-07:
the script is not detecting the firmware type correctly, sigh. Google doing stupid shit on these older devices when booted in RW_LEGACY. what's the output of `sudo dmidecode -t 0`

Vany913, 2024-11-08:
it doesn't seem to work :(

MrChromebox, 2024-11-08:
I don't know what that means

Vany913, 2024-11-08:
here I enter the command and then the result
[image]
[image]

MrChromebox, 2024-11-08:
the problem is Google deleted the version field when booting RW_LEGACY, so I can't tell what firmware you are running.

exit the script and run these commands:
`cd /tmp`
`sudo ./gbb_utility --set --flags=0xa9 bios.bin`
`sudo ./flashrom -p internal -w bios.bin --fmap -i GBB -N`

if that works, you can reboot and it should auto boot to Linux

Vany913, 2024-11-08:
after loading it stops there
[image]

MrChromebox, 2024-11-08:
sorry, the flags should have been 0x4a9, so manually boot with CTRL+L *before* it boots to recovery, then re-run the script / re-run the commands above with the correct flags value

Vany913, 2024-11-08:
The recovery mode screen appeared again with part of the command line, but after that Linux still booted. If this is how it should be, then that’s it, all the problems regarding rw_legacy seem to have been resolved. I am grateful to you :)

MrChromebox, 2024-11-08:
that's not the recovery screen ("ChromeOS is missing or damaged"), that's the Developer Mode boot screen ("OS verification is off").

Vany913, 2024-11-08:
I meant developer mode, sorry.

MrChromebox, 2024-11-08:
right, that's the expected boot flow with the stock firmware. Only other thing you can do is strip out the bitmaps so that it's a black screen vs white

Vany913, 2024-11-08:
ok, thanks again. I think we can close the question ;)

closed 2024-11-08 by Vany913