MrChromebox/scripts #431: Возврат к стандартной прошивке с помощью скрипта не возможен на pixel book go

Mir-Win, 2024-10-28:
Делаю по сценарию но стандартная прошивка удалена, прошу помощи.
[image]
[image]
[image]
[image]
[image]

Mir-Win, 2024-10-28:
Я хотел бы вернуть стоковую прошивку и использовать Chrome OS  как это было задумано.

Mir-Win, 2024-10-28:
[image]

Mir-Win, 2024-10-28:
If you don't mind, give me a link to what needs to be done, I'll do it

MrChromebox, 2024-10-28:
please attach the file /tmp/flashrom.log here so I can see it

Mir-Win, 2024-10-28:
[Uploading flashrom.log…]()

Mir-Win, 2024-10-29:
Good afternoon, I want to clarify, did I do what I needed to do?

MrChromebox, 2024-10-29:
It's unclear to me from the log why you're running into this issue, need to ask someone else for input

Mir-Win, 2024-10-29:
Сan you recommend a specialist in this matter?

MrChromebox, 2024-10-29:
I'm talking to the folks on #flashrom in IRC, they're the experts

Mir-Win, 2024-10-30:
Hello dear friend, this could not be due to the fact that the disk is crookedly broken?
sda               8:0    1  14,3G  0 disk  
├─sda1            8:1    1    10M  0 part  
├─sda2            8:2    1    16M  0 part  
├─sda3            8:3    1   2,3G  0 part  
├─sda4            8:4    1    16M  0 part  
├─sda5            8:5    1     2M  0 part  
├─sda6            8:6    1   512B  0 part  
├─sda7            8:7    1   512B  0 part  
├─sda8            8:8    1    16M  0 part  
├─sda9            8:9    1   512B  0 part  
├─sda10           8:10   1   512B  0 part  
├─sda11           8:11   1     8M  0 part  
└─sda12           8:12   1    32M  0 part  
mmcblk0         179:0    0 115,2G  0 disk  
├─mmcblk0p1     179:1    0 263,1M  0 part  /boot/efi
├─mmcblk0p2     179:2    0 976,6M  0 part  /boot
└─mmcblk0p3     179:3    0   114G  0 part  
  └─cryptdata   252:0    0   114G  0 crypt
    ├─data-root 252:1    0 110,2G  0 lvm   /
    └─data-swap 252:2    0   3,8G  0 lvm   [SWAP]
mmcblk0boot0    179:8    0     4M  1 disk  
mmcblk0boot1    179:16   0     4M  1 disk

MrChromebox, 2024-10-30:
I don't see any issue with the disk partition structure here -- what is your concern?

Mir-Win, 2024-10-30:
I think there are different options, could there be a problem with the emmc flash chip?

MrChromebox, 2024-10-30:
the eMMC would not affect the flashing of the firmware, that is on a separate chip

Mir-Win, 2024-10-30:
Maybe it's possible to do this through the programmer?

Mir-Win, 2024-10-30:
By the way, I tried to disconnect the battery, connect the power and do it again according to the script, but I got the same result

Mir-Win, 2024-10-30:
this is what official [support](https://support.google.com/chromebook/thread/304202598/%D0%92%D0%BE%D1%81%D1%81%D1%82%D0%B0%D0%BD%D0%BE%D0%B2%D0%BB%D0%B5%D0%BD%D0%B8%D0%B5-chrome-os?hl=en)

Mir-Win, 2024-10-30:
you can also think about a chip spi flash with bootloader, what do you think about this?

MrChromebox, 2024-10-30:
> ![Снимок экрана 2024-10-30 164337](https://private-user-images.githubusercontent.com/53466484/381575468-78455239-5323-4c21-81ed-43040ab9b091.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAzMDAwMTgsIm5iZiI6MTczMDI5OTcxOCwicGF0aCI6Ii81MzQ2NjQ4NC8zODE1NzU0NjgtNzg0NTUyMzktNTMyMy00YzIxLTgxZWQtNDMwNDBhYjliMDkxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDMwVDE0NDgzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI3MzRlZGYyMmJhMmJiZTgzNmUzMmViNTlmMjc3NGQ0MWJhYzA4NmJlZjc4NTdhNTNkMGI3NmRjNmNkYWU2NjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.u90IzqedULwWOWImFwfNQ_A_buvu3Yzs_rLluKhdyhc) this is what official [support](https://support.google.com/chromebook/thread/304202598/%D0%92%D0%BE%D1%81%D1%81%D1%82%D0%B0%D0%BD%D0%BE%D0%B2%D0%BB%D0%B5%D0%BD%D0%B8%D0%B5-chrome-os?hl=en)

that person does not know what they are talking about

> you can also think about a chip spi flash with bootloader, what do you think about this?

you can of course do this, but the flash chip is likely not clippable (WSON-8 package, not SOIC-8), so would require a different set of tools to flash externally

Mir-Win, 2024-10-30:
Here's what you managed to get to
[image]

Mir-Win, 2024-10-30:
[image]

Mir-Win, 2024-10-30:
Но после попытки восстановления перезагрузка и все начинается заново...

MrChromebox, 2024-10-30:
you cannot use a ChromeOS Recovery USB with the MrChromebox firmware, you have to restore the stock firmware first

Mir-Win, 2024-10-30:
Great, can you give advice on how to do this?

MrChromebox, 2024-10-31:
> I followed the instructions

not based on the picture posted.

You need to restore the stock firmware using my script. That's not working, and what prompted you to open this issue. You cannot do anything else until that issue is resolved. Recovery media is useless without stock ChromeOS firmware.

Mir-Win, 2024-10-31:
OK, then what can you try to restore the firmware using your script? Could I make a mistake in the process somewhere?

Mir-Win, 2024-10-31:
Maybe I need to do some actions for you so that you can better understand what the problem is?

Mir-Win, 2024-10-31:
I also want to understand for myself whether your script can download the firmware from the Internet on its own, I read that it is possible [here](https://www.howtogeek.com/279308/how-to-restore-your-chromebooks-original-bios-and-software/)

MrChromebox, 2024-10-31:
> I also want to understand for myself whether your script can download the firmware from the Internet on its own, I read that it is possible [here](https://www.howtogeek.com/279308/how-to-restore-your-chromebooks-original-bios-and-software/)

that article is from 2017, the feature was removed some time ago. I don't have the time to maintain a catalog of stock firmware images for every device I support just because people can't be bothered to safely store a backup of their firmware

Mir-Win, 2024-11-01:
Thanks for your honest answer! Now everything is clear to me and there is no need to think that I did something wrong (well, except for not taking care to save the standard firmware).
But I do not lose hope that you can still help me and give me an image of the standard firmware, perhaps not now, but soon, and I will do what I wanted.
Thank you again for your time, perhaps the image you provided will help someone else, and this post will save people from rash actions when flashing their device

Mir-Win, 2024-11-01:
By the way, I bought another laptop with the same firmware. I can take her image using your script? and then stop your pixel?

MrChromebox, 2024-11-01:
Here is the latest recovery image firmware for ATLAS. If you use the firmware from your other device, then they will have the same serial # which might post problems. Also, you cannot read the entire flash image from a live system (though you can with an external programmer).

[coreboot-Google_Atlas.11827.198.0.zip](https://github.com/user-attachments/files/17600544/coreboot-Google_Atlas.11827.198.0.zip)

Mir-Win, 2024-11-01:
Thank you very much for your efforts. Tomorrow I will continue the experiments and describe the results obtained.

Mir-Win, 2024-11-02:
Dear friend, I tried different recovery methods, but they didn’t work either and I can’t figure out the reason. Can I continue to try to use your help and once again post the whole story here step by step for observation and control on your part?

MrChromebox, 2024-11-02:
> Dear friend, I tried different recovery methods, but they didn’t work either and I can’t figure out the reason. Can I continue to try to use your help and once again post the whole story here step by step for observation and control on your part?

not sure what you mean, what did you try?

flashing the firmware I provided above using a USB programmer is the best way to proceed at this point for reverting to ChromeOS

Mir-Win, 2024-11-02:
Теперь стало еще понятнее,  спасибо! Но таких знаний у меня точно нет, буду искать специалиста, который мне поможет в этом, по результатам, отпишусь.

closed 2024-11-02 by Mir-Win