MrChromebox/scripts #432: Acer CB514 (Manuf. on 07/24) not allowing RW Legacy, nor full ROM (UEFI) installation

cutervino, 2024-11-01:
This new chromebook, doesn't recognize the micro sd card, when trying to boot the new OS I installed on it (RW Legacy). Also, when I tried to flash UEFI, following he instrucions for disabling WP, it doesn't turn on when charger is connected (Battery disconnected) to try to write mr chromebox scrypt.  Of course, when this chromebook is not on developer mode, it turns on normally. Any help?

MrChromebox, 2024-11-01:
without the ChromeOS board name as shown in the script menu, I can't do anything to help

cutervino, 2024-11-01:
CRASSNETO-KYVC

Sent with [Proton Mail](https://proton.me/mail/home) secure email.

MrChromebox, 2024-11-01:
SD card not being bootable is a known issue with the edk2 UEFI payload used for RW_LEGACY and affects many devices.

The WP type / disablement method for CRASSNETO on the supported devices page does **NOT** include battery disconnect, so not sure why you are trying to do so. Using a SuzyQ cable is your only option.

closed 2024-11-01 by MrChromebox