MrChromebox/scripts #433: Script not starting on Falcon HP 14 CrOS 76

Fortato, 2024-11-02:
The script doesn't do anything, there are 0 errors and 0 messages appearing. I have previously dual booted Ubuntu but factory restarted since. I'm now trying to fully replace CrOS with arch but am having some trouble with this script. Any help is appreciated
[image]

MrChromebox, 2024-11-02:
sounds like curl is hanging for some reason. try using `-LOv` for verbose output, or `-LOk` to ignore SSL certificate issues (which are likely on a ChromeOS version that old).

I would have told you just to flash from Ubuntu FWIW =D

Fortato, 2024-11-02:
i tried both and im having the same hanging issue, not sure where to move from here

MrChromebox, 2024-11-02:
are your time/date correct? have you tried running from a VT2 terminal?

Fortato, 2024-11-02:
time/date seem good. vt2 terminal also wasnt working

MrChromebox, 2024-11-02:
FALCO should have functional RW_LEGACY OOTB, can you boot a live Linux USB and run the script from there?

Fortato, 2024-11-03:
i used esc + refresh + power to get to the recovery screen, but when attempting to boot i get "the device you inserted does not contain chrome os", i tried that ctrl + l then after boot esc but nothing happens and after it stops beeping i return to a login screen.

MrChromebox, 2024-11-03:
> i used esc + refresh + power to get to the recovery screen, but when attempting to boot i get "the device you inserted does not contain chrome os",

well yes, booting to recovery mode only allows you to boot recovery media -- that's how it works.

power on the device normally, and hit CTRL+L on the developer mode boot screen ("OS verification is off").
if it beeps, then boot ChromeOS and enable legacy boot mode by running the following in a shell: `sudo crossystem dev_boot_legacy=1` then reboot and try CTRL+L again

Fortato, 2024-11-03:
oh that worked thanks, ill try running your script now

Fortato, 2024-11-05:
So i finally got it working, ended up just reflashing chromeos after i accidentally broke it (oops!) but after the reinstall it worked perfectly fine. Thank for the help mr chromebox!

closed 2024-11-05 by Fortato