MrChromebox/scripts #434: Can't restore stock firmware Acer CB311-10H

Junkob101, 2024-11-07:
Hello,

I'm having issues restoring the stock firmware onto my Chromebook using the recovery media for the firmware.
The device is an Acer CB311-10H (board name KASUMI) running Fedora 40 and I'm using recovery media from the Acer C721. 

The restore process seems to work up until it needs to grab the firmware from the USB drive, in which it errors out in the copying process. 
I could be using the wrong recovery media but the C721 was the closest match I could find based off of the board name and specifications. 

Any assistance/pointers would be greatly appreciated :) 

[Firmware script log](https://github.com/user-attachments/files/17657857/chromebox_firmware_script.log)

MrChromebox, 2024-11-07:
as the script log shows, it was unable to extract the firmware from the recovery USB for some reason:

> cp: target 'Google_Grunt.11031.123.0.bin': No such file or directory
> mv: cannot stat 'coreboot-Google_*': No such file or directory
> ERROR: Unable to open /tmp/stock-firmware.rom for reading: No such file or directory

so I need to look at see 1) why that failed, and 2) why the script didn't catch the error and halt

Junkob101, 2024-11-07:
Works perfectly now. Thank you very much for your help!

closed 2024-11-07 by Junkob101

ynoalberto, 2024-11-08:
Hello

I have this error when I want to restore stock firmware using a ChromeOS Recovery USB in fedora 41:

cp: -r not specified; omitting directory '/tmp/tmp.QT9rIcvgVJ/'
Error: failed to extract firmware image for blorb using this recovery USB

MrChromebox, 2024-11-08:
@ynoalberto if not for the same device, then please open a new ticket

ynoalberto, 2024-11-08:
No. I open a section in issusse. Check it please

ynoalberto, 2024-11-08:
In that section I explains the device that I have