MrChromebox/scripts #435: Can't return to stock firmware on acer chromebook CB315 BLORB

ynoalberto, 2024-11-08:
Hello

I have this error when I want to restore stock firmware in my acer chromebook CB315 BLORB using a ChromeOS Recovery USB in fedora 41:

cp: -r not specified; omitting directory '/tmp/tmp.QT9rIcvgVJ/'
Error: failed to extract firmware image for blorb using this recovery USB

MrChromebox, 2024-11-08:
hmm, I just tested the extraction function on a blorb recovery image and it worked correctly. Are you sure the recovery USB is correct for blorb?

ynoalberto, 2024-11-08:
Yes. I maje the recovery image U.S. Inf de chrome os utility. Have you any stock-firmware.rom file for my chromebook? Or Need I try with another USB?

MrChromebox, 2024-11-08:
you can use the crosfirmware.sh script from https://github.com/coreboot/coreboot/tree/main/util/chromeos to do the same thing the firmware-util script does - downloads a recovery image and extracts the firmware for it. I tested it for blorb and no issues.

ynoalberto, 2024-11-08:
I have this issue when extract the voreboot images. If you have this file for my mothderboard, send me here: 

[image]

ynoalberto, 2024-11-08:
I don't know what's happends. You says don't have any issues, but I have some issues anda I'm sure that my board is BLORB

MrChromebox, 2024-11-08:
I fixed the problem in the firmware-util.sh script the other day and in croshfirmware.sh just now.

this is the correct image for BLORB, though it does not have a valid HWID or correct GBB flags.
[coreboot-Google_Bobba.11297.357.0.zip](https://github.com/user-attachments/files/17683160/coreboot-Google_Bobba.11297.357.0.zip)

ynoalberto, 2024-11-08:
Sorry. In the croshfirmware.sh I don't look any change on the github. But I will to try the file that you send me

MrChromebox, 2024-11-08:
I just pushed a fix to the firmware-util. script. copy/paste error from the coreboot change

ynoalberto, 2024-11-08:
All works fine. Thanks so much. Sorry for wasting your time.

MrChromebox, 2024-11-08:
no worries, there was an issue with the script so good it got fixed

closed 2024-11-08 by MrChromebox